billw2/rpi-clone #9: Please add -x to rsync options

te36, 2015-01-07:
I hd a bunch of NFS mounted file systems on my RPI and rpi-clone happily cloned them into the clone root filesystem. This can easily be avoided by addin -x to the RSYNC options. This should also allow you to simplify the program because you should not need any 'exclude' anymore for RSYNC (right now there is a long list) - all the exemptions are additional mounted filesystems that will not be copied anymore with '-x'.

andig, 2015-02-26:
I guess this has been done and issue can be closed?

SpoddyCoder, 2017-05-27:
@te36 @andig this has been resolved by the 1.6 release. Please download and install latest rpi-clone. If this resolves your issue, please click close, thanks :)