billw2/rpi-clone #12: Bill...  Are you out there?  Need NOOBS support

Jeppedy, 2015-03-19:
Bill, NOOBS is something that is becoming more and more prevalent.  It seems that rpi-clone is not compatible with a NOOBS card.  I imagine the whole issue is that NOOBS has more partitions which rpi-clone is not set to handle.

For someone familiar with Linux file systems, the change should be fairly easy.  There are just more partitions to create/copy.  But to those not completely confident, there's a lot of room for error :-)

Can you please help or at least give some guidance to those willing to try modifying the scrip to manage it?

scargill, 2015-04-06:
I believe I installed from a NOOBS setup so this would be really useful.

Jeppedy, 2015-04-06:
I didn't understand Linux partitioning and file system usage. But I did some research and it made sense. Basically:
- create two partitions on the new SD card 
- 'cp' the contents of each partition, the boot and the root partitions, to the new card
- CRITICAL!  Edit the boot files and fstab to reflect the new partition numbers. (Other references to this process on the web don't mention this)

All Done!

(See elsewhere on web for specifics for each step. I'd recommend creating a fresh non-NOOBS card for a clean reference as you do this.)

samighi, 2015-04-10:
If I understand it, the original and this copy both copy only two partitions. I have tried to modify it for NOOBS, but it is more complicated due to size issues. 4gb to 8gb or 8gb to 4gb rsync process.  

The NOOBSs installs 3 or more partitions. If I remember right it creates 6 partions. 

I am surprised no one has updated this for NOOBS. I use this for raspian all the time. It is the only way to ensure accurate copies while the system is operational. Also, keeps refreshing easy.

jproehl, 2015-04-16:
So I'm puzzled. I installed Raspian from the NOOBS distribution and chose to try the rpi-clone route to duplicate the filesystem. I did so, because I wanted to get the image off of a 16Gb MicroSD card and put it on an 8Gb one.  Will this script work in its current form?  

I actually started the script before posting this message and I am wondering how long it should take for an 868mb filesystem?  It's been working for about 3 hours + and I have no idea if it's hung or just grinding along.  I started with a blank (just purchased) MicroSD card and got the warning about needing to initialize the disk (/dev/sda1), which I told it to do. The last message printed was:

Imaging the partition structure, copying 868 megabytes ...

Any way to tell if it's hung or working??

Thanks!

samighi, 2015-04-16:
See other posts. rpi-clone doesn’t work work with NOOBS. Having said that, there are several other factors that would take a long time. 

rpi-clone uses rsync. using control+t might give you a status like dd does. 
it only excludes sda sdb and a few others, so if you have a lot of drives attached, it will try to copy them also. I had to exclude sde, sdf for example. Change this line 
OPTIONAL_MNT_DIRS="clone mnt sda sdb rpi0 rpi1"
the other factor is your card reader or your SD card. i have seen a factor of 4x 6x difference between SD class 4 and class 10+. Also some card readers are super slow. 20 min+ for 4 GB, but 3 hours for your file seems like a lot. 
Since it uses rsync, unless it has bad blocks on the destination, I am not sure what else could be wrong. if you cancel (control-c) add v to this list, you will see what it is doing

SYNC_OPTIONS="--force -rltWDEgopt"

SpoddyCoder, 2017-06-19:
@Jeppedy @scargill @samighi not intended as a competitor - I hope to keep contributing to rpi-clone core functionality - but I've just released a new tool called ClonePi tool which supports NOOBS multi-partition setups...

https://github.com/SpoddyCoder/clonepi

samighi, 2017-06-19:
Excellent. I failed at my attempt w parted years ago and was wondering why it wasn't done by an expert already :) 

Thank you. I have used the script for years and will use the new one again. It was amazing. And I hope the new one as successful. 

I use it heavily in replicating my class work. Doing image copies is very painful for minor changes. 

Saman

https://github.com/SpoddyCoder/clonepi