billw2/rpi-clone #13: Automatically clone without user interaction

onkelandy, 2015-06-18:
He there! Thanks for the very nice script. Are you planning to include an additional flag that makes it possible to run the script as a crontab. I mean without having to confirm anything.. but logging the output in a specific logfile? That would be great! Thanks!

crowzero, 2015-06-18:
Dear onkelandy,
I added a quiet mode (-q) to the script. A pull request was started right now. I use this quite mode now since March 2015 as a cron job. 

```
 Version 1.3	2015/06/18
	* add -q (quiet) option for use as cron script
         -q only works with existing backups, 
         it won't initialize the destination SD card partitions with dd!
         Please initialize the destination SD card from the shell (by hand)
         for the first time.
         This is to protect you for accidentally overwriting a SD Card.
```

onkelandy, 2015-06-19:
Awesome, crowzero - thanks alot! Seems to work fine!

closed 2015-08-10 by onkelandy

jdrch, 2021-05-03:
Just switched to using this after my `-U` crontab entry was failing.