billw2/rpi-clone #19: Orange Pi

scargill, 2015-10-19:
Hmm-  trying this on an orange Pi which has a 32gig SD. I wanted to copy to a 16gig SD. I formatted the 16 gig to FAT32 and ran the script. All went well for a moment.. Any ideas?

The existing destination disk 'sda' partitions are:
Disk /dev/sda: 16013MB
Partition Table: loop

Number  Start   End      Size     File system  Flags
 1      0.00MB  16013MB  16013MB  fat32

*** All data on destination disk sda will be overwritten! ***

Do you want to initialize the destination disk /dev/sda? (yes/no): yes

Imaging the partition structure, copying 96 megabytes...
96+0 records in
96+0 records out
100663296 bytes (101 MB) copied, 4.88343 s, 20.6 MB/s
Running fsck on /dev/sda1...
Sizing partition 2 (root partition) to use all SD card space...

Do you really want to quit? mke2fs 1.42.12 (29-Aug-2014)
The file /dev/sda2 does not exist and no size was specified.

/dev/sda is initialized and resized.  Its partitions are:
Error: Can't have a partition outside the disk!

djsuszi, 2017-07-26:
It's probably because Raspberry PI card use 2 partition on card, VFAT and Linux.
Orange PI (Raspbian) has only one Linux Partition.

@scargill Have you got working script to make backup for sdcard from on orange pi?

scargill, 2017-07-26:
Hi there

 

No – I don’t – however I did have some luck with the standard rsync…. Managed to get a second board up and running… didn’t make notes though…. Erm, isn’t the first pi partition FAT32?