billw2/rpi-clone #23: Add -w option to leave wiggle room?

GTR2Fan, 2016-02-27:
Hi Bill,

Thanks for this excellent script. It's much appreciated. I've written a guide on how to install it under Raspbian with a desktop GUI launcher as a sign of my appreciation here... https://www.raspberrypi.org/forums/viewtopic.php?p=914109#p914109

I was wondering if it might be possible for the /root partition to be optionally slightly undersized so that a clone created on one, let's say, 4GB card could be guaranteed to fit on to any other 4GB card.

As I'm sure you're aware, exact available capacity varies slightly between cards of different makes/models/batches even if they're still marketed as being xGB in size, so adding a -w option to allowing trimming of maybe 100MB would give enough wiggle-room for a "fits-all" solution to this problem.

Maybe make the exact amount of wiggle-room adjustable by the end-user specified in MB, for example, '-w 150' for 150MB of wiggle room?

Could an additional small partition be created at the end of the card to reserve this space, then deleted at the end of the cloning process?

I don't even know if this is possible as my knowledge of Linux partition handling is close to zero, but it would be a very useful addition to the tool IMO if this is doable.

Kind regards, GTR2Fan.

billw2, 2016-03-17:
Thanks for that GUI guide.  I've added a link to it in the README.md.

Since rpi-clone works when cloning to cards of different sizes I'm not sure that having a -w option
adds any real benefit to using rpi-clone.   Well, I guess unless you want to use rpi-clone to have a
way to have the SD card more easily backed up by some other method... We'll see.

fjpdevries, 2016-06-22:
This feature is just what i am looking for!

Once I've finished developing, I'd like to make a master image file and save that on my PC
For new installations I'd like to take take this master image and write it to a new target SD card instead of creating a nwe image form the pi itself.