billw2/rpi-clone #24: Operation not permitted

gruenberg, 2016-03-01:
I have an issue when trying to clone to an empty sd card.
The script hangs at (see below) and its subfolders/files
***
Starting the filesystem rsync to sda
(This may take several minutes)...rsync: chgrp "/mnt/clone/boot" failed: Operation not permitted (1)
...
***

Any help is appreciated!

billw2, 2016-03-17:
It might be pretty verbose, but could you post the output when using the -x option to rpi-clone?
Maybe just attach the -x output as a file.

arigit, 2017-01-04:
FWIW - I also got  rsync errors on 'chown' to files in the "mnt/clone/boot/" partition. The errors were reproducible on any attempt, even after a reboot (and removal and re-insertion of the backup Card)

Since that partition is FAT - I think chown errors can be safely ignored; the boot partition is FAT and doesn't support file owner attributes.  Proof: the system booted perfectly on the new card (fully configured media center).

Curiously, after rerunning rpi-clone in the same scenario but in debug mode: 
rpi-clone -x  
the errors didn't show up, even in subsequent runs with -x  (even without -x after that run).
Not sure if this was related to the "-x" or could have been related to timing (wait longer after boot before reattempting the cloning) or else

SpoddyCoder, 2017-05-27:
Hi @gruenberg @arigit  - I think this issue should be resolved by the 1.6 release, please download & install latest rpi-clone. If this fixes your issue, please click close thanks! :)

krismc, 2017-06-05:
I've downloaded the latest version but still getting a page of errors like
rsync: chown "/mnt/clone/boot/." failed: Operation not permitted (1)
rsync: chown "/mnt/clone/boot/COPYING.linux" failed: Operation not permitted (1)
rsync: chown "/mnt/clone/boot/LICENCE.broadcom" failed: Operation not permitted (1)
rsync: chown "/mnt/clone/boot/LICENSE.oracle" failed: Operation not permitted (1)
rsync: chown "/mnt/clone/boot/bcm2708-rpi-b-plus.dtb" failed: Operation not permitted (1)
rsync: chown "/mnt/clone/boot/bcm2708-rpi-b.dtb" failed: Operation not permitted (1)
rsync: chown "/mnt/clone/boot/bcm2708-rpi-cm.dtb" failed: Operation not permitted (1)
rsync: chown "/mnt/clone/boot/bcm2709-rpi-2-b.dtb" failed: Operation not permitted (1)
rsync: chown "/mnt/clone/boot/bcm2710-rpi-3-b.dtb" failed: Operation not permitted (1)
rsync: chown "/mnt/clone/boot/bcm2710-rpi-cm3.dtb" failed: Operation not permitted (1)
rsync: chown "/mnt/clone/boot/bootcode.bin" failed: Operation not permitted (1)

SpoddyCoder, 2017-06-05:
HI @krismc - we'll need more info on what's happening on your system as I'm unable to reproduce this. Please post the following details;

1) Your OS type

Before rpi-clone runs....

2) Output of: `sudo fdisk -l`
3) Output of: `sudo df -h`

After rpi-clone runs, but BEFORE you tell rpiu-clone to unmount the SD card (you'll need a 2nd shell window to do this)...

4) Output of: `sudo df -h`
5) Output of: `sudo ls -al /mnt/clone/`
6) Output of: `sudo ls -al /mnt/clone/boot/`

krismc, 2017-06-06:
On Tue, Jun 6, 2017 at 1:23 AM, Paul Fernihough <notifications@github.com>
wrote:

> HI @krismc <https://github.com/krismc> - we'll need more info on what's
> happening on your system as I'm unable to reproduce this. Please post the
> following details;
>
>    1. Your OS type
>
>
>    1. root@raspberrypi:/home/pi# uname -a
>
>
>    1.
>
>    Linux raspberrypi 4.4.38-v7+ #938 SMP Thu Dec 15 15:22:21 GMT 2016 armv7l GNU/Linux
>
>
> Before rpi-clone runs....
>
>    1. Output of: sudo fdisk -l
>
> Disk /dev/ram0: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disk /dev/ram1: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disk /dev/ram2: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disk /dev/ram3: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disk /dev/ram4: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disk /dev/ram5: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disk /dev/ram6: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disk /dev/ram7: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disk /dev/ram8: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disk /dev/ram9: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disk /dev/ram10: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disk /dev/ram11: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
[... 101 lines cut ...]
> -rwxr-xr-x  1 pi   pi     15440 Jun  5 23:44 bcm2708-rpi-b-plus.dtb
> -rwxr-xr-x  1 pi   pi     14900 Jun  5 23:44 bcm2708-rpi-cm.dtb
> -rwxr-xr-x  1 pi   pi     16507 Jun  5 23:44 bcm2709-rpi-2-b.dtb
> -rwxr-xr-x  1 pi   pi     17608 Jun  5 13:41 bcm2710-rpi-3-b.dtb
> -rwxr-xr-x  1 pi   pi     16364 Jun  5 13:41 bcm2710-rpi-cm3.dtb
> -rwxr-xr-x  1 pi   pi     50268 Jun  5 23:44 bootcode.bin
> -rwxr-xr-x  1 pi   pi       153 Dec 24 23:21 cmdline.txt
> -rwxr-xr-x  1 pi   pi      1596 Jun  5 13:41 config.txt
> -rwxr-xr-x  1 pi   pi     18693 Jun  5 23:44 COPYING.linux
> -rwxr-xr-x  1 pi   pi      2578 Jun  5 23:44 fixup_cd.dat
> -rwxr-xr-x  1 pi   pi      6660 Jun  5 23:44 fixup.dat
> -rwxr-xr-x  1 pi   pi      9799 Jun  5 13:41 fixup_db.dat
> -rwxr-xr-x  1 pi   pi      9795 Jun  5 13:41 fixup_x.dat
> -rwxr-xr-x  1 pi   pi       145 May 27  2016 issue.txt
> -rwxr-xr-x  1 pi   pi   4576784 Jun  5 23:44 kernel7.img
> -rwxr-xr-x  1 pi   pi   4376232 Jun  5 13:41 kernel.img
> -rwxr-xr-x  1 pi   pi      1494 Jun  5 23:44 LICENCE.broadcom
> -rwxr-xr-x  1 pi   pi     18974 May 27  2016 LICENSE.oracle
> drwxr-xr-x  2 pi   pi     16384 Jun  5 13:42 overlays
> -rwxr-xr-x  1 pi   pi    656868 Jun  5 13:41 start_cd.elf
> -rwxr-xr-x  1 pi   pi   4989092 Jun  5 23:44 start_db.elf
> -rwxr-xr-x  1 pi   pi   2851012 Jun  5 13:41 start.elf
> -rwxr-xr-x  1 pi   pi   3934596 Jun  5 23:44 start_x.elf
>
>

SpoddyCoder, 2017-06-06:
Hi @krismc 

At what point do you insert your SD card? It looks like your OS is mounting it to an incorrect location. From the output I can see the destination drive (clone) is mounted like so...
```
/dev/sda1        56M   22M   35M  39% /home/pi/usbdrv
/dev/sda2       7.2G  1.2G  5.7G  17% /mnt/clone
```
Whereas it should be mounted like...
```
/dev/sda1        56M   22M   35M  39% /mnt/clone/boot
/dev/sda2       7.2G  1.2G  5.7G  17% /mnt/clone
```
The mount to `/home/pi/usbdrv` is not being done by rpi-clone, so it must be getting mounted by your OS. Try disabling auto-mounting of USB cards in your OS. Or you can do the following after you have inserted the card and allowed a minute for the OS to auto-mount...
```
sudo umount /home/pi/usbdrv
``
And then run rpi-clone. Let us know how you get on.

mysiki, 2021-12-27:
I confirm that @SpoddyCoder say (in my case) problem was that the system auto mount drive during process. 

For my case, I try to clone osmc OS, need to run following command before rpi-clone

```
sudo systemctl disable udisks-glue
sudo systemctl disable udisks
```

Additional : In order to full clean disk before copy, I use wipefs command.