billw2/rpi-clone #26: feature request: please make rpi-clone also work for NOOBS-based Raspbian (Jessie)

shiftleftplusone, 2016-05-11:
hello,
i have a feature request: 
please make rpi-clone also work for NOOBS-based Raspbian (Jessie) SD cards
thank you very much in advance!

fochs, 2016-09-16:
a second voice for this support -- I just acquired a raspberry pi 3 and thescript does not work; the NOOBS sD card (16gb) contains 

/dev/mmcblk0:16,1GB:sd/mmc:512:512:msdos:SD USD:;
1:4194kB:1172MB:1168MB:fat32::lba;
2:1172MB:15,8GB:14,6GB:::;
5:1174MB:1208MB:33,6MB:ext4::;
6:1208MB:1274MB:66,1MB:fat16::lba;
7:1275MB:15,8GB:14,5GB:ext4::;

and the mounted partitions are #6 and 7:
/dev/mmcblk0p6  /boot           vfat    defaults          0       2
/dev/mmcblk0p7  /               ext4    defaults,noatime  0       1

shiftleftplusone, 2016-09-23:
since the new Raspi kernel has been released in April, there is eventually a SD cloning tool in the menu bar which works fine even for cloning images on different sized SD cards (as long as it wouldn't run out of memory) and it works currently very satisfactorily. So raspi-clone is actually not needed so very urgently any more.

fochs, 2016-09-23:
thank you for the news, the version I have is dated Feb.25 2016, just a bit too old... Anyway with the mixture of dd and rsync commands
detailed in your script, it was not too difficult to realize the cloning...

Thanks for your help !

SpoddyCoder, 2017-06-19:
@shiftleftplusone @fochs not intended as a competitor - I hope to keep contributing to rpi-clone core functionality - but I've just released a new tool called ClonePi tool which supports NOOBS multi-partition setups...

https://github.com/SpoddyCoder/clonepi

shiftleftplusone, 2017-06-19:
thank you, but since the latest Jessie releases have that great partition cloning tool in the launch panel I'm absolutely fine with this feature, both for bigger target SDs and even when down-cloning from 32GB to 16GB SD cards.

SpoddyCoder, 2017-06-19:
@shiftleftplusone thanks for the feedback - I've developed it primarily to run on headless Pi's, so didn't even know this existed in Pixel.... *runs off to test it out*