billw2/rpi-clone #28: rpi-clone crosses filesystems inappropriately

chrissv, 2016-07-06:
I had a very early version of rpi-clone that worked fine. I upgraded to the latest, and I noticed the removal of the "x" option in RSYNC_OPTIONS causes problems.

My RPi has an external HDD mounted (for NAS-type backups). Without the "x" option, rpi-clone attempts to include my 3TB HDD on the 16GB SD card clone - obviously this doesn't work.

Was there rationale for removing the "x" option for rsync? I can't think if a case where you'd want the clone to cross filesystem boundaries.

I temporarily fixed my problem by adding back in "x" to RSYNC_OPTIONS.

Bombdogs, 2017-05-21:
Quick note to say that I was expecting to fall foul of this (3TB mounted ntfs drive for a plex-server) but didn't - process worked flawlessly. So this issue might be system/setup specific... @chrissv might be worth posting more detail on your particular setup and the rpi-clone output.

chrissv, 2017-05-22:
Here is some detail for my system:

```
root@xxx:~# df
Filesystem      1K-blocks       Used  Available Use% Mounted on
/dev/root        15186900    3561308   10958256  25% /
devtmpfs           469532          0     469532   0% /dev
tmpfs               94776       1380      93396   2% /run
tmpfs                5120          0       5120   0% /run/lock
tmpfs              189540          0     189540   0% /run/shm
/dev/mmcblk0p1      57288      21720      35568  38% /boot
/dev/sdc1      2930264060 1674459784 1255804276  58% /toshiba
none               473864          0     473864   0% /sys/fs/cgroup

root@xxx:~# mount
/dev/root on / type ext4 (rw,noatime,data=ordered)
devtmpfs on /dev type devtmpfs (rw,relatime,size=469532k,nr_inodes=117383,mode=755)
tmpfs on /run type tmpfs (rw,nosuid,noexec,relatime,size=94776k,mode=755)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
proc on /proc type proc (rw,nosuid,nodev,noexec,relatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /run/shm type tmpfs (rw,nosuid,nodev,noexec,relatime,size=189540k)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
/dev/mmcblk0p1 on /boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
fusectl on /sys/fs/fuse/connections type fusectl (rw,relatime)
/dev/sdc1 on /toshiba type fuseblk (rw,nosuid,nodev,relatime,user_id=0,group_id=0,allow_other,blksize=4096)
none on /sys/fs/cgroup type tmpfs (rw,relatime)
none on /sys/fs/cgroup/memory type cgroup (rw,relatime,memory)
```

The problem for me was the `/toshiba` mount point - rpi-clone tried to include that in the clone operation without the 'x' rsync option added.

Bombdogs, 2017-05-22:
Ahha moment of clarity - my media drive is mounted to /mnt/plex-library and /mnt/ has been specifically excluded in the rsync options.... because that's where the clone is stored (line 450).

I'd agree that it's very rare that you'd ever want this to cross the filesystem boundary, so I'd +1 re-addition of the -x option to the rsync options to be a bit more friendly to varied setups.

@billw2 perhaps make this an option if you think it's necessary to be included? Happy to submit a pull request if you take them?

billw2, 2017-05-22:
It needs to cross the /boot filesystem boundary, so that's why no -x.  To have that the script
would have to be modified to separately rsync to /boot.  Or maybe there could be an option
to add a "--include pattern", but offhand I don't know how it behaves if the include pattern falls
inside an excluded directory.  That would need to be tested.

But if you do come up with a fix for what you need, I can take a pull request or a patch.

Bill

chrissv, 2017-05-22:
An alternate solution might be a command line argument (or ENV variable) of additional mount points to ignore. It could be added to the existing list of ignoring in the script.

SpoddyCoder, 2017-05-24:
@chrissv @billw2 This is my personal account - previously posts were under work alias. I've just submitted a pull request with some improved cross filesystem handling. 

Bill - I think your first instinct was correct - split the rsync into two separate processes. I added a --cross-filesystem switch for those wishing to maintain existing behaviour - tho I suspect usage of this will be rare, so made no cross filesystem default behaviour - hope it makes the cut :)

billw2, 2017-05-25:
Thanks, I'll have to give it a workout.  But it might be a day or two before I can
get to it as I'm tied up with some other other things right now.

Bill

SpoddyCoder, 2017-05-25:
Yep understood @billw2 - of course I tested a range of use cases before submitting the pull request too :)

@chrissv you may also help out here with the regression testing - you can grab new version here...

https://raw.githubusercontent.com/SpoddyCoder/rpi-clone/a37954c93020b74247ee9546b856b6c53d6e48ba/rpi-clone

And try it out for size on your system (take a backup first) - please let us know if you hit any issues.

chrissv, 2017-05-26:
Just gave it a try, and it seemed to work. I enabled -x and -v and watched it, and it skipped my /toshiba mount as planned.

Thanks - looks good.

billw2, 2017-05-27:
All looks good and I've merged your no cross filesystem changes.  Thanks for taking
care of the details of updating the changelog and help message.

I've gone ahead and removed the /mnt and  /media excludes.

chrissv, 2017-05-27:
Side question: what is the purpose of line 500? :
```
rm -f $CLONE/etc/udev/rules.d/70-persistent-net.rules
```

SpoddyCoder, 2017-05-27:
This is created at boot time so any clone will automatically generate it. @billw2 will know the removal reason better than me - but I suspect it's because it will contain MAC address info which is not ideal to have in place on actual clones (different machines) - tho this is no problem for a  plain backup (same machine). 

@billw2 thanks for accepting the pull request and happy to contribute. Also nice to see the /mnt /media excludes go 👍 

@chrissv nice to see it working on your system - you can now close this issue up :)

closed 2017-05-27 by chrissv

billw2, 2017-05-27:
On Fri, 26 May 2017 18:23:06 -0700
Paul Fernihough <notifications@github.com> wrote:

> This is created at boot time so any clone will automatically generate it. @billw2 will know the removal
> reason better than me - but I suspect it's because it will contain MAC address info which is [...]

That's right.  Maybe it's not an issue anymore because I don't see 70-persistent-net.rules
being generated anymore on my Pi's.  Or maybe I did something on my system to stop it from
being generated.

Anyway, if the rule was there and a cloned file system was moved to a different machine, its net
devices would show up as say wlan1 or eth1 instead of the wanted wlan0 or eth0 because
the rule was causing the wlan0/eth0 names to be reserved for the device that was on the
machine the clone was sourced from.