billw2/rpi-clone #32: Read-only file?

robustini, 2016-11-04:
I try the latest version of your script and i've a lot of error during the cloning:

Starting the filesystem rsync to sda
(This may take several minutes)...rsync: write failed on "/mnt/clone/usr/bin/git-upload-pack": Read-only file system (30)

Another if i use "-f" option:

rsync: chown "/mnt/clone/boot/bootcode.bin" failed: Operation not permitted (1)

rsync error: received SIGINT, SIGTERM, or SIGHUP (code 20) at rsync.c(632) [sender=3.1.1]
rsync error: received SIGINT, SIGTERM, or SIGHUP (code 20) at io.c(504) [generator=3.1.1]
./rpi-clone: line 494: /mnt/clone//var/log/rpi-clone.log: No such file or directory

robustini, 2016-11-05:
For help you in the debug this version of script work fine:

https://www.raspberrypi.org/forums/download/file.php?id=16104&sid=63a32c66c0d9a5d29e60d5f3712518c4

robustini, 2016-11-05:
For help you in a possible debug this simply manual step work fine here:

http://sysmatt.blogspot.it/2014/08/backup-restore-customize-and-clone-your.html

SpoddyCoder, 2017-05-27:
Hi @robustini - this issue should be resolved by the 1.6 release, please download & install latest rpi-clone. If this fixes your issue, please click close thanks! :)