billw2/rpi-clone #42: Allow cloning from sda1 instead of mmblk0 for USB-booting setups

TheDiveO, 2017-08-17:
Some of my Raspberry Pis are set up to boot from an USB mass storage device, so they don't have any mSD card inserted. I would like to clone and backup these systems too, without having to resort to the GUI-based clone program. Can you please add a command line option that allows changing the source block device?

SpoddyCoder, 2017-08-17:
This is quite a big change, so it may be a while before it's made. The source device is referenced at several points in the rpiclone script and has some implicit assumptions about partition identifiers - which may be different for your device.

ClonePi which performs a similar function to rpiclone, but extends it in a number of ways to be more flexible can be found here...

https://github.com/SpoddyCoder/clonepi

... this allows the block device and partition identifiers can be specified in conf. I haven't tested your specific use case, but it should work.

framps, 2017-08-17:
Support for the new Raspi3 feature you're asking for, to boot from USB only, is available in [raspiBackup](https://github.com/framps/raspiBackup). Just give it a try.

TheDiveO, 2017-08-17:
While I can somehow understand people advertising their solutions, I would like to ask commenters here to refrain from further advertising. My issue is directed at rpi-clone which works just fine, thank you.

To bring this issue discussion to a sensible level: the number of places where `mmcblk0` is used is limited. For partitions it's the stem `mmcblk0p` plus partition number or stem `sda` plus partition number. No rocket science here. Nothing that could be handled in the form of _devicestem_+_partdesig_+_partno_. For mSD: _devicestem_ `mmcblk0`, _partdesig_ `p`; while for other block device names the _partdesig_ will be simply empty "".

framps, 2017-08-18:
Looks like you didn't follow the link: raspiBackup is OpenSource and free as rpi-clone is. I thought I do you a favor pointing to raspiBackup which supports the feature you asked for. Instead you want to use rpi-clone and wait for somebody to add this support to rpi-clone. Maybe this guy can grab some code from raspiBackup to speed up the implementation. It took me a while to get this feature implemented and in particular tested.

Frankly I wouldn't have created raspiBackup if I would have known about this famous rpi-clone when I bought my first Raspi.

@billw2 Please let me know if you have an issue with my comments and I'll delete them.

SpoddyCoder, 2017-08-18:
@TheDiveO  Pffft - I'm a contributer to rpi-clone too. As @framps says - there are other free tools out there that can do what you need now. Pardon me for trying to be helpful...

Look at the pull request history & commit history for rpi-clone and you will find that it moves quite slowly, you'll likely be waiting a while for this feature to be added.

billw2, 2017-08-19:
**@TheDiveO**  I've removed the hardwired mmcblk0 and rpi-clone should now work from your booted USB disk.  The booted disk should be automatically determined from /etc/mtab, but if that doesn't work for some reason, there is a command line option for source disk as you mentioned.  Hopefully you won't need to use that and running rpi-clone will just work with usage as before.

TheDiveO, 2017-08-19:
@billw2 Many thanks for the quick response, this is highly appreciated. So we can use the bug tracker for tracking issues, not as a discussion forum. ;)

closed 2017-08-19 by TheDiveO

TheDiveO, 2017-08-21:
Tried the updated script today on a Pi 3B and it works like a charm. Thank you very much!