billw2/rpi-clone #45: suggestions

fragolinux, 2017-08-31:
1st, very well done script!
2nd, some suggestions for enhancements

- add an option to clone to an usb connected drive, useful for those who have boot fs on sd and root one on hd or usb dongle, for example...
- add an option for unattended cloning, no pause at the end waiting for mods...
- add an option to do a SYNC backup, so adding the --delete switch to rsync to remove stuff from DST sd that's not anymore on SRC sd...

thanks in advante

scargill, 2017-08-31:
Yes  -u for unattended would be good for me...

fragolinux, 2017-08-31:
@billw2 we added 2 of the suggestions to this fork, in case you want to pull them back to main repo:
https://github.com/scargill/rpi-clone/blob/master/rpi-clone

billw2, 2017-09-01:
On Thu, 31 Aug 2017 16:13:29 +0000 (UTC)
fragolinux <notifications@github.com> wrote:

> @billw2 we added 2 of the suggestions to this fork, in case you want to pull them back to main repo:
> https://github.com/scargill/rpi-clone/blob/master/rpi-clone
> 

Thanks, I'll have them added along with other changes I'm currently working on for the
next release.

Bill

scargill, 2017-09-01:
Thanks - just to let you know - in total I have 3 items  

1 - optional change of hostname - one of the guys has submitted this but I think only updated HOSTNAMES - wheres I normally update both HOSTS and HOSTNAMES.
2  -  -d DELETE option - which I think is essential for cloning - if you need an exact copy.
3 -  -u for unattended...

HOWEVER right now (hence I'm writing as it isn't out there) - I had an issue - I have a NEO2 which has it's ROOT partion on a hard drive (SDA1).

So I altered your 	SRC_ROOT to say SDA1

And it worked - I have successfully backed up an SD+HD Neo2 to an SD - popped the new SD in and it all works - the ONLY issue I'm having is that the SD (which I cleared completely - removing all partitions) is a 32GB SD  and for some reason - the script has only created a 14GB partition for ROOT -  I'd have expected it to use up all the space. We'll look at this but if you know better PLEASE do tell.   If all of this goes into the next version I'll happily scrap my fork - no point in having two separate versions....

Thoughts on the disk size (the original SD was 16G - the hard disk holing the root partition was 56GB yet my new 32GB SD ends up with the boot partition and a 14GB root partition....

billw2, 2017-09-01:
On Fri, 01 Sep 2017 18:39:35 +0000 (UTC)
Peter Scargill <notifications@github.com> wrote:

> Thanks - just to let you know - in total I have 3 items
>
> 1 - optional change of hostname - one of the guys has submitted this but I think only updated HOSTNAMES -
> wheres I normally update both [...]

1 - There will be ways to set the hostname:  by option or by script.   2 - rpi-clone has
always been using --delete.  It's hardwired at the spot where the rsync command is
run, so that option is redundant, unless I'm missing something? 
3 - I've added in your unattended.

> HOWEVER right now (hence I'm writing as it isn't out there) - I had an issue - I have a NEO2 which has it's
> ROOT partion on a hard drive (SDA1).
> 
> So I altered your 	SRC_ROOT to say SDA1

The new version will automatically figure out the root partition number and it won't\
be hardwired to partition 2.  Please test that when I get the next release out.
 
> And it worked - I have successfully backed up an SD+HD Neo2 to an SD - popped the new SD in and it all works
> - the ONLY issue I'm having is that the SD (which I cleared completely - removing all [...]

Only the last partition can be resized to fill the disk.  So if your root is partition 1 you will
only get the same size as the source if there are more partitions.  I'll look at if there can
be an option to pick the partition to resize.  How is that disk organized, is boot partition 2?
Can you show me Neo2 output from these commands:

sudo parted print
sudo parted -m print
findmnt -o source,target,fstype -l | grep "^/dev/"

Also, how does the Neo2 boot?  Is just files in a /boot directory or does a bootloader
like grub (or what?) need to be installed.  rpi-clone will have an alternate way to
clone if a boot loader can be separately installed like grub.  In this case a destination
disk can be prepared manually with partition sizes you want (and even different numbers
of partitions) as long as partitions and file systems are created on the destination to match
the booted disks mounted partitions.  So far I've only been able to test this on my Debian
desktop that uses grub.

scargill, 2017-09-02:
Hi

Ok, we missed DELETE and so added it. I've just been told it is there. Oh, well, only wasted 10 minutes :-)

Hostname - we only added it as I update both hosts and hostname files and only one was implemented in the other fork.. Good to know...

Partitions - GREAT - looks like I can relax and wait for update... 

You asked for NEO output - here it is (I've gone back to the original setup, not the SD I made... 

First first request - sudo parted print...

Error: Could not stat device print - No such file or directory.

Second one returned exactly the same.

Third one returns:

/dev/sda1      /                               ext4
/dev/mmcblk0p1 /boot                 vfat

As for Neo booting - there is a fat boot partition - and it has a file called boot.scr - there is a text version called boot.cmd which you alter the name of the partition if you've set up for hard disk - and you recompile that file...

These are the files I can see in the boot partition. I've attached an image.

[image]

If you need more info or want me to test something - my email is pete at scargill dot org - by all means get in touch - I've a variety of board - NEO2 NEO PLUS2 - OODROID C2 etc which I can test.

Good work - I'm used to the SD backup on the Pi and it is sorely missed elsewhere so these changes are most welcome.

scargill, 2017-09-02:
Antonio suggested I give you this info... the output from these 4 commands

sudo parted /dev/sda -m print
sudo parted /dev/sda print
sudo parted /dev/mmcblk0 -m print
sudo parted /dev/mmcblk0 print

Here they are in order...

BYT;
/dev/sda:60.0GB:scsi:512:4096:msdos:JMicron Tech:;
1:1049kB:60.0GB:60.0GB:ext4::;

------------

Model: JMicron Tech (scsi)
Disk /dev/sda: 60.0GB
Sector size (logical/physical): 512B/4096B
Partition Table: msdos
Disk Flags:

Number  Start   End     Size    Type     File system  Flags
 1      1049kB  60.0GB  60.0GB  primary  ext4

-----------------

1:25.2MB:130MB:105MB:fat16::;
2:130MB:16.0GB:15.9GB:ext4::;

--------------------

Model: SD 00000 (sd/mmc)
Disk /dev/mmcblk0: 16.0GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags:

Number  Start   End     Size    Type     File system  Flags
 1      25.2MB  130MB   105MB   primary  fat16
 2      130MB   16.0GB  15.9GB  primary  ext4

billw2, 2017-09-02:
On Sat, 02 Sep 2017 09:56:25 +0000 (UTC)
Peter Scargill <notifications@github.com> wrote:

> Antonio suggested I give you this info... the output from these 4 commands
> 
> sudo parted /dev/sda -m print
> sudo parted /dev/sda print
> sudo parted /dev/mmcblk0 -m print
> sudo parted /dev/mmcblk0 print
> 
> Here they are in order...

That info was what I needed, thanks.

But one more thing, does the Neo2 use PARTUUID in /etc/fstab and /boot/cmdline.txt
like recent Raspbians do?  Or does it just use device names like previous Raspbians.
Maybe if I could I see those files (or the Neo2 equivalent of cmdline.txt if any)?

I want to be sure of what to expect so I can cover my bases for the various use cases
of rpi-clone I want to work.

scargill, 2017-09-02:
FSTAB contains

/dev/mmcblk0p1    /boot vfat defaults 0 0

There is no /biit.cmdline.txt - just the file boot.cmd which compiles to boot.scr. Here is the content of my boot.cmd

# Recompile with:
# mkimage -C none -A arm -T script -d boot.cmd boot.scr

setenv fsck.repair yes
setenv ramdisk rootfs.cpio.gz
setenv kernel Image

setenv env_addr 0x45000000
setenv kernel_addr 0x46000000
setenv ramdisk_addr 0x47000000
setenv dtb_addr 0x48000000

fatload mmc 0 ${kernel_addr} ${kernel}
fatload mmc 0 ${ramdisk_addr} ${ramdisk}
fatload mmc 0 ${dtb_addr} sun50i-h5-${board}.dtb
fdt addr ${dtb_addr}

# setup MAC address 
fdt set ethernet0 local-mac-address ${mac_node}

# setup boot_device
fdt set mmc${boot_mmc} boot_device <1>

setenv bootargs earlyprintk root=/dev/sda1 rootfstype=ext4 rw rootwait fsck.repair=${fsck.repair} panic=10 ${extra}
booti ${kernel_addr} ${ramdisk_addr}:500000 ${dtb_addr}

scargill, 2017-09-02:
Anything else I can get?

billw2, 2017-09-02:
On Sat, 02 Sep 2017 13:06:53 -0700
Peter Scargill <notifications@github.com> wrote:

> Anything else I can get?

That's good for me to get done what I want to do.  Now it's just a case of your
giving it a test run when I get the next version out.  If any issues pop up then it
shouldn't be a problem to deal with them.

scargill, 2017-09-02:
Can't wait....

billw2, 2017-09-22:
Should be resolved in new version 2.0

closed 2017-09-22 by billw2