billw2/rpi-clone #47: No -b option to findmnt on wheeze raspbian

6b6561, 2017-09-26:
rpi-clone fails on wheeze as there is no -b option available on findmnt command.

`root@xxx:/ss/rpi-clone# rpi-clone /dev/sda
findmnt: invalid option -- 'b'

Usage:
 findmnt [options]
 findmnt [options] <device> | <mountpoint>
 findmnt [options] <device> <mountpoint>
 findmnt [options] [--source <device>] [--target <mountpoint>]

Options:
 -s, --fstab            search in static table of filesystems
 -m, --mtab             search in table of mounted filesystems
 -k, --kernel           search in kernel table of mounted
                          filesystems (default)

 -p, --poll[=<list>]    monitor changes in table of mounted filesystems
 -w, --timeout <num>    upper limit in milliseconds that --poll will block

 -a, --ascii            use ASCII chars for tree formatting
 -c, --canonicalize     canonicalize printed paths
`

billw2, 2017-09-26:
The -b option is removed.
Thanks,

closed 2017-09-26 by billw2

6b6561, 2017-09-27:
Confirmed to work! A big thank you!