billw2/rpi-clone #49: can't copy SD card on raspberry pi anymore

lebo202002, 2017-10-19:
when I type in the following commands...
cd rpi-clone

sudo cp rpi-clone /usr/local/sbin
ls -la /dev/disk/by-id
look for the last sd_ on your sd card reader
sudo rpi-clone sda -v
or
sudo rpi-clone sdb -v
or 
sudo rpi-clone sdc -v
or
sudo rpi-clone sdd -v
answer y for unmount
answer y for initialize the destination disk /dev/sd_ 

I get the following issue... 

Initializing
  Imaging past the start of /boot partition 2.
  => dd if=/dev/mmcblk0 of=/dev/sdb bs=1M count=65 ...
  Resizing last partition to end of disk ...
    Resize failed.  See /tmp/rpi-clone-output.
    Try running rpi-clone again.

Re-reading the partition table failed.: Device or resource busy
pi@retropie:~/rpi-clone $

and it does not copy.. what am I doing wrong?!

TheDiveO, 2017-10-24:
I see the same issue on Raspbian Jessie with the recent rpi-clone update: since the **cloned** mSD on `sda` has the same part IDs as the mounted boot mSD, rpi-clone (or maybe some of the tools it uses) thinks that `sda` mSD is actually mounted on "/", while it isn't mounted, as `df` shows: `/dev/mmcblk0p1` is still mounted as `/boot` and `/dev/root` as `/`.

TheDiveO, 2017-10-24:
More details: I selectively moved backwards along master until 2d8412dd12fec63260b678f307b43891872fc953, which is the last working 1.x version. From 2.0 on, `rpi-clone` fails as it thinks that the backup mSD is already mounted on `/`.

TheDiveO, 2017-10-24:
@lebo202002 while cd'ed into your git clone of the github rpi-clone repo, do `git checkout 2d8412d`. This winds you back to rpi-clone 1.x. Then run `sudo ./rpi-clone sda`.

hoRacy, 2017-12-11:
Brillant! Thanks! 2d8412d works for me on cloning retropie image to smaller sd card!

niveki, 2018-10-14:
Hello, i have the same issu with the version 2.0 (the last) on my raspberry pi 3 I want to clone my SDD on a SD card. error : `Initializing
  Imaging past the start of /boot partition 2.
  => dd if=/dev/sda of=/dev/mmcblk0 bs=1M count=50 ...
  Resizing last partition to end of disk ...
    Resize failed.  See /tmp/rpi-clone-output.
    Try running rpi-clone again.` 
i had tyr to put the code 2d8412d but he don't work : error : `Final check, is it Ok to proceed with the clone (yes/no)?: yes
=> Mounting /dev/mmcblk0p12 (no label) on /mnt/clone
mount: special device /dev/mmcblk0p12 does not exist
Mount failure of /dev/mmcblk0p12, aborting!`

How I can do to clone my SSD ? 

sorry for my bad English I'm French ^^
thank for you futur answer