billw2/rpi-clone #52: Can't get unattended backup to write to log

Lorccan, 2018-01-04:
I am using cron to schedule backups and it's working fine.  What I am unclear on is how to get it to write to a log (or, indeed, how the log file is designated).

I've tried using -q and -u options, but I must be missing something.

Could anyone tell me please:

a) where is the log file defined (or is it just by redirecting output)?
b) what command should I use in my crontab to run unattended but also write out the log to this file?

larrym, 2018-05-28:
1 0 * * * /usr/local/sbin/rpi-clone -f -U >> /root/rpi-clone.log 2>&1

CCurvin, 2020-02-11:
The help output (from running rpi-clone with no parameters) says:

>  A line logging a rpi-clone run is written to /var/log/rpi-clone.log.

Lorccan, 2020-02-25:
> The help output (from running rpi-clone with no parameters) says:
> 
> > A line logging a rpi-clone run is written to /var/log/rpi-clone.log.

Thank you. Don't know how I missed that!

closed 2020-02-25 by Lorccan