billw2/rpi-clone #56: Impossibile unmount sda2 from /

sherazan, 2018-02-23:
Hi, I'm using with success the first versio of RPI-Clone on a Raspberry with raspbian Linux rpi3mh 4.9.46-v7+ #1032 SMP Wed Aug 30 12:09:14 BST 2017 armv7l GNU/Linux and every synch work fine.

After installing V2 of RPI-Clone, and executing "sudo rpi-clone sda -v", I receive the following message:

  **_Destination disk partition /dev/sda2 is mounted on /.
  The clone cannot proceed unless it is unmounted.
Do you want to unmount /?  (yes/no): yes
umount: /: target is busy
        (In some cases useful info about processes that
         use the device is found by lsof(8) or fuser(1).)
rpi-clone could not unmount /.
Aborting!_**

The source and destination are the same...

Where is the problem?

Please help me.

Thanks

Guido

SeeSpotRun, 2018-03-08:
> sudo rpi-clone sda -v

This command will clone to sda...

> Destination disk partition /dev/sda2 is mounted on /.

... but sda is the disk that you booted from.  In other words you are trying to clone sda to sda which is a bad idea.  You need to either clone to a different disk or boot from a different disk,

TheDiveO, 2018-06-05:
That's not necessarily meaning that sherazan booted from sda. rpi-clone 2.0 is broken since many months,  see issue #49. Fix is to roll back to last known working 1.x, see https://github.com/billw2/rpi-clone/issues/49#issuecomment-338959704

robustini, 2018-12-16:
Same here, i roll back to 1.7.1 and all work fine.
When you could fix the 2.x version?