billw2/rpi-clone #60: Error: Can't have a partition outside the disk!

mrgw454, 2018-07-13:
Hello.  Benn using this great utility for a while on many SD cards (64GB or less).  Tried my first 128GB card today and received this error.  I have to CTRL-C to exit.  Any idea on what would cause this?  Thanks!

smrfeld, 2019-04-01:
Same problem!

trevjonez, 2019-07-08:
Same issue here, thinking it is due to the source disk being larger than the destination so partition map of the source doesn't fit within destination space. I'm going to give a spin at shrinking the partition on the source first

trevjonez, 2019-07-08:
I believe a bad partition table on the destination is what is triggering the issue as I was getting the same issue just trying to list partitions in gparted.

Here is what I did to get around it. 
1. Used `cfdisk` to delete all the partitions from destination drive.
2. Used `cfdisk` to reduce the partition size of the source drive partition such that they fit within the destination drive ***dangerous depending on where data is written?
3. rerun the rpi-clone command.

My clone command is still running but seems as though it is going to work.