billw2/rpi-clone #63: Hard drive backups.

scargill, 2018-10-24:
Not an issue, more a dumb question, actually 2. I'm using RPI-clone on RPI3 every day and  you've put me off using other boards that may not work with this, it is so useful. I maintain a distrust of SDs even though I can now easily back them up (when I'm there). I spend 6 months in the UK and 6 months in Spain and I need my kit to work when I'm not there. I use:

sudo pi-clone -U sda

I'm backing up from sd to sd in a usb adaptor and it works every time.

Q1: If I change to booting and running from a hard drive (USB2, assuming I can find split leads that will let this happen power wise..) is the process the same for backing up from that to another USB hard drive?

Also is there an option to change IP address as part of the process?

Thanks for your work on this sterling software.

Pete

scargill, 2019-07-11:
I continue to sing the praises of rpi-clone  as I use it daily on my RPI3 SD setups. I did not manage to get a response however on HD use. As I'm moving to RPI4, can anyone advise. I currently use "sudo rpi-clone -U sda" unless it's a new SD in which case I use "sudo rpi-clone -f -U sda"

Imagine a scenario where FIRSTLY one might move the entire installation to hard disk - and THEN  looking ahead, copy disk to disk.  What is the least change to my current routine to achieve this?

dmurphydrtc, 2019-09-25:
Did you find a solution to this problem? I'm in the same boat. Thanks in advance.