billw2/rpi-clone #64: Errors sudo crontab rpi-clone sda -u

thunder2k, 2018-11-05:
Hi,
Thanks for this great project, that's exactly what I was looking for! It's working flawlessly when I run it manually.

When I tried to schedule the following command via sudo crontab:
`/usr/local/sbin/rpi-clone sda -u`
I got the following error output via email:

> Command not found: parted      Package required: parted
> Command not found: fdisk       Package required: util-linux
> Command not found: fsck.vfat   Package required: dosfstools
> 
> Do you want to apt-get install the packages?  (yes/no): Aborting!

Am I missing something or there's an issue here?

Thanks!!

charlie-young, 2019-04-03:
I was also having this same issue and discovered that the script has no PATH set and thus the cron user is unable to properly locate those commands. By editing the script with the following it fixed it:

vi /usr/local/sbin/rpi-clone

add the following under the first line of "#!/bin/bash":
#!/bin/bash
export PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/root/bin"

thunder2k, 2019-04-03:
Thanks for the hint, I'll test it and let you know!

heylookltsme, 2020-04-25:
Many thanks @charlie-young! I was having this exact issue as well 👍

jdrch, 2021-05-10:
> #!/bin/bash
> export PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/root/bin"

Oddly enough, although I have the latest version of the script on my Pi 3B+, this line was not present. I added it. Hopefully I stop getting that error message.