billw2/rpi-clone #66: Clone without unmounting 3rd partition

jonferreira, 2019-01-22:
Noob here to excuse me if it's a stupid question.

I'm following steps (7) to backup my RPI into a USB flash drive with 3 partitions: boot, root and data (which is supposed to be left untouched)

I've followed the instructions to initialise the USB so that it matches the SD card

[image]

Used fdisk to create the partitions, set the right type and then

mkfs -t vfat -F 32 /dev/sda1
mkfs.ext4 /dev/sda2
mkfs.ext4 /dev/sda3

While it "works" OK while I don't have sdb3 mounted, if I do have it mount (which I'll need to) then it fails immediately with 

"Destination disk partition /dev/sdb3 is mounted on /mnt/tmp.
  The clone cannot proceed unless it is unmounted."

My understanding was that sdb3 was supposed to be left alone?

jonferreira, 2019-01-24:
OK I've edit the script as follows to only check for mount on partition 1 and 2

`dst_mount_table=$(findmnt -o source,target -n -l \
				| grep "^/dev/${dst_disk}1\|^/dev/${dst_disk}2" | tr -s " ")`

tyjtyj, 2020-02-10:
This should consider as fix. There is not need to check for sdX3. I having same issue. Thanks for the temporary fix