billw2/rpi-clone #70: Don't work on osmc

fmarzocca, 2019-03-14:
I am trying to use the script with osmc (www.osmc.tv) when installing a dependency (bsdmainutils) I got an error: `Error: Could not stat device /dev/systemd `

Any tip?

arigit, 2019-08-24:
Same issue here. Did you ever find a solution?
I also have two OSMC installs, one with rpi-clone working and the other without.
However, both are at the same version (july 2019) , the only difference is that one is a fresh install vs. the other one is an old install that has been upgraded over time.
The difference  must be some package missing in the fresh install that rpi clone depends on and is not documented  (my guess)

arigit, 2019-08-24:
I found the issue. 

In the OSMC fresh install, rpi-clone fails to determine the source disk, which rpi-clone tries to guess by finding which device hosts the /boot directory, by using findmnt.

This is strange thing: in the fresh install,

```
mount |grep boot
systemd-1 on /boot type autofs (rw,relatime,fd=30,pgrp=1,timeout=0,minproto=5,maxproto=5,direct)
/dev/mmcblk0p1 on /boot type vfat (rw,noatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
```

There is the strange first line  showing "systemd-1". This is also reflected by the command that rpi-clone uses:

```
/home/osmc/rpi-clone# findmnt /boot -o source -n
systemd-1
/dev/mmcblk0p1
```

So the quick hack to solve this is:
look for the following line:
```
src_boot_dev=`findmnt /boot -o source -n`
```

and replace it with
```
src_boot_dev=`findmnt /boot -o source -n | grep -v -e "systemd"`
```

After this rpi-clone worked perfectly in my system

fmarzocca, 2019-08-25:
Great! Thanks you very much!!

fmarzocca, 2019-08-26:
Yes, you are right!

billw2, 2019-08-26:
On Mon, 26 Aug 2019 10:28:23 -0700
Ari <notifications@github.com> wrote:

> @billw2 Yes - this works, and it's a better approach. That way you are guaranteed to catch always the
> physical device which is the one you need for rpi-clone

Thanks for the feedback.  I've pushed the change to filter findmnt results for /dev/

fmarzocca, 2019-08-27:
> > src_boot_dev=`findmnt /boot -o source -n | grep  "/dev/"`

I confirm this is working on my osmc too...