billw2/rpi-clone #73: Feature request - stop services before cloning and (re)start after

mrWheel, 2019-05-15:
Hi,

To prevent changing files during the cloning process it would be nice if you stop services like "cron", "mysql", "mail" etc.

Before cloning:
```
systemctl stop mysql.service
systemctl stop cron.service
```
-- and others --

After cloning is done:
```
systemclt start cron.service
systemctl start mysql.service
```

fmarzocca, 2019-06-20:
I agree with this.

BTW: which is the right command line syntax for a crontab? -q or -u?

framps, 2019-06-20:
It's a very good idea to stop all services updating or caching data before starting the backup of a running system. :+1: 

There is an easy workaround: Just follow the [Linux philosophy of having every tool doing it's job best](https://en.wikipedia.org/wiki/Unix_philosophy)  and combine them either with pipes or other means. Said that, just write a small wrapper script which will stop all relevant services first and then calls rpi-clone. At the end start all the services you stopped previously in reverse sequence.

I spent some time to find an algorithm to identify and stop all important services in my backup tool but finally gave up. You have to have to configure the sequence of services to stop and start manually. systemd doesn't help you on this - unfortunately :disappointed:

CCurvin, 2020-02-11:
I've been attempting this and found this page to be helpful:
https://www.tecmint.com/list-all-running-services-under-systemd-in-linux/

`systemctl --type=service --state=running`

Then add that to something like an awk string:
`systemctl --type=service --state=running | grep ".service" | awk '{print $1}'`

And then something like a for loop:
`for i in $(systemctl --type=service --state=running | grep ".service" | awk '{print $1}') ; do systemctl stop $i ; done`

But in order to be able to start those services back up after, maybe redirect to a file, then throw it into a for loop:
`systemctl --type=service --state=running | grep ".service" | awk '{print $1}' > /tmp/rpi-clone.services`
`for i in $(cat /tmp/rpi-clone.services); do systemctl stop $i ; done`
Perform rpi-clone, then:
`for i in $(cat /tmp/rpi-clone.services); do systemctl start $i ; done`

Something along those lines. I was typing this out while testing out the various options. If I end up throwing it into a script, I'll reply back with more info (assuming it's successful of course ;) ).

framps, 2020-02-11:
I use a similar logic in raspiBackup :wink:

CCurvin, 2020-02-11:
Now I'm going to have to go find raspiBackup and check it out too! Thanks for the hard work and for the tools!

framps, 2020-02-11:
@CCurvin Just wrap rpi-clone in a bash script and add pre and post steps to stop and start services using your code if you're familiar with bash :smiley: