billw2/rpi-clone #75: New Out of Space issues with Buster

scargill, 2019-07-24:
Hi

Over more than a year I've used RPI-CLONE without issue and now I'm on Buster on RPI 3, 3B+ and 4 after initially backing up all of them, I'm having major issues backing from SD to SD or SD to USB.

Here's a typical example...

Initializing
  Imaging past the start of /boot partition 2.
  => dd if=/dev/mmcblk0 of=/dev/sda bs=1M count=52 ...
  Resizing last partition to end of disk ...
    Resize success.
  Changing destination Disk ID ...
  Delaying so partprobe can update /dev entries ...
  => fsck -p /dev/sda1 ...
  => mkfs -t ext4  /dev/sda2 ...

Syncing file systems (can take a long time)
Syncing mounted partitions:
  Mounting /dev/sda2 on /mnt/clone
  => rsync // /mnt/clone with-root-excludes ...
  Mounting /dev/sda1 on /mnt/clone/boot
  => rsync /boot/ /mnt/clone/boot  ...rsync: write failed on "/mnt/clone/boot/kernel7l.img": No space left on dev                           ice (28)
rsync error: error in file IO (code 11) at receiver.c(374) [receiver=3.1.3]

Editing /mnt/clone/boot/cmdline.txt PARTUUID to use 8bd6e6a0
Editing /mnt/clone/etc/fstab PARTUUID to use 8bd6e6a0
===============================
Done with clone to /dev/sda
   Start - 14:54:09    End - 15:20:50    Elapsed Time - 26:41
  unmounting /mnt/clone/boot
  unmounting /mnt/clone
===============================

Why would there be an issue on the boot partition? I'm using as usual

sudo rpi-clone -U sda
or if that fails
sudo rpi-clone -f -U sda

Again nothing new there?

xjnxjnxjn, 2019-07-31:
Hi,

I ran into the same issue.

The basic reason is, that the default Debian Stretch /boot partition size is only about 40M and
with Debian Buster update 93% of the volume is used (The default Debian Buster /boot partition size is about 128M).

---

The reason with rpi-clone is, that the "rsync" command in "rpi-clone" uses a "temporary copy file" on the TARGET partition and the target partition has not enough space left for that.

To solve this issue maybe the rysnc's "--temp-dir=DIR" option could be used to point to a local tmp-space, delegated as command-line-parameter  ...

Or create a complete new backup to an unformatted device ...

br
xjn

scargill, 2019-07-31:
Bill - are you able  to factor this into rpi-clone? I'm not an expert on this - and would rather not make a custom version - after all, Buster IS the new official Raspbian so rpi-clone needs to work reliably as it always has in the past...  

XJN - if you get any further insights, do suggest them to Bill - and by all means let me know at https://tech.scargill.net/raspberry-pi-buster-issue/  as I've been trying to cludge my way around this problem - I'm not entirely happy with my results....

scargill, 2019-07-31:
XJN - I just took a look at rpi-clone script and can't even make a start at understanding where the boot partition size is defined. Any thoughts?

xjnxjnxjn, 2019-07-31:
Hi,

may be I was not precise enough.

The "... No space left on device ..." error only occurs if one tries to "rpi-clone" to a previous "Debian Stretch" rpi-clone backup partition, after a "Debian Buster" update. 

The (older) default "Debian Stretch" partition is too small to hold the additional "tmp-file" with Buster update ...

---

The easiest way is to use a new clean unformatted target disk for the "Buster" backup, 
Then everythig seems to be fine, at lest for me.

br
xjn

scargill, 2019-08-01:
XJN - thanks for that... in that case, If I do a clone using -f -U options, then that SHOULD repartition the SD - that's the point of the -f option surely... yet I ALSO in that case have to reboot the PI  FIRST....  So... right now I have my upgraded PI with it's 16GB SD and  Buster on it.. and in the last couple of days it backed up successfuly to another 16GB SD which is now again plugged into the USB socket (this was done using reboot then -f -U - after the upgrade) - so all should be well. I've done nothing except adding a tiny temperature test to my Node-Red.  Right now I will do a check with CAT /PROC/PARTITIONS.

pi@raspberry-spain:~:08:38[0]> cat /proc/partitions
major minor  #blocks  name

   1        0       4096 ram0
   1        1       4096 ram1
   1        2       4096 ram2
   1        3       4096 ram3
   1        4       4096 ram4
   1        5       4096 ram5
   1        6       4096 ram6
   1        7       4096 ram7
   1        8       4096 ram8
   1        9       4096 ram9
   1       10       4096 ram10
   1       11       4096 ram11
   1       12       4096 ram12
   1       13       4096 ram13
   1       14       4096 ram14
   1       15       4096 ram15
 179        0   15558144 mmcblk0
 179        1      44236 mmcblk0p1
 179        2   15508992 mmcblk0p2
   8        0   15558144 sda
   8        1      44236 sda1
   8        2   15508992 sda2

All should be well and a simple -U option on RPI-clone (no reboot, no -f option) SHOULD work... and in a very short space of time too...  and SURE ENOUGH - IT JUST WORKED.

So then I took my 32GB SD which 2 days ago I used to make a backup with from the same upgraded original - and tried again without  a reboot - or -f options (i.e. only using -U) - and again - it WORKED.  I've just done the same with my 32GB USB stick - and again it WORKED.

Perhaps you are onto something - or maybe I just woke up in a parallel universe this morning.

Short of using rpi-clone - how do you manage to recreate a "clean, unformatted disk from a used one?

scargill, 2019-08-03:
AH BUT, in making my new clone and subsequent clones of that, I've JUST realised that RPI-CLONE, despite being able to FORCE recreation of partitions, still isnt aware that the BUSTER BOOT partition needs to be largere. My clones WORK but there is still no extra room for future cloning of clones given the increased requirements of BUSTER. So, somehow on that first clone, we (I cannot be th ONLY one cloniing upgrades) need to be able to force the target to adopt a larger BOOT partition and accordingly slightly smaller ROOT partition befor starting the copy process... anyone any idea how to do this? my  SD partitions are 1024 bytes long (don;t ask why, I don't know) and the BOOT partition (remember this is FAT16) in both original and current clones is 44236 blocks in size). The ROOT partition (EXT4) is 15558144 blocks but of course not ALL is used.  Thoughts? Something tells me that apart from initially moving a lot of data around (hands free) this should be simple....

billw2, 2019-08-25:
I added a "-p size" option to rpi-clone so a clone can resize /boot.  If you clone to a previously cloned SD card you will have to force initialize with -f.  If you forget the -f, at least rpi-clone will show a warning if your source partition is too small.  The new Buster /boot is 256M.

scargill, 2019-08-26:
Excellent Bill - I did this manually using GPARTED but probably got it wrong - so to see this added is great. Will test today.

SolarDuck, 2019-09-04:
How about using the rsync --inplace option?  Any issues?

scargill, 2019-09-04:
Bill's -p option sems perfect - I'm using it on a daily basis to update from older setups.