billw2/rpi-clone #78: Read-only filesystem (30)

darrepac, 2019-08-09:
Hi

I just downloaded rpi-clone and tried it.
It fails always with a rsync error like:
`rsync: write failed on "/mnt/clone/usr/lib/arm-linux-gnueabihf/dri/vmwgfx_dri.so": Read-only file system (30)`

(not always at the same file)

If it helps, here is the message at start:
`
Booted disk: mmcblk0 16.0GB                Destination disk: sda 8.0GB
---------------------------------------------------------------------------
Part      Size    FS     Label           Part   Size    FS     Label
1 /boot   58.4MB  fat16  --              1      58.4MB  fat16  --
2 root     8.0GB  ext4   --              2       7.9GB  ext4   backupdomo
---------------------------------------------------------------------------
== SYNC mmcblk0 file systems to sda ==
/boot                 (24.1MB used)  : SYNC to sda1 (58.4MB size)
/                     (7.1GB used)   : SYNC to sda2 (7.9GB size)
---------------------------------------------------------------------------
Run setup script       : no
Verbose mode           : no
-----------------------:
`

then when it happens, I can't try anymore: 
`Syncing mounted partitions:
  Mounting /dev/sda2 on /mnt/clone
mount: /dev/sda2: can't read superblock
    Mount failure of /dev/sda2 on /mnt/clone.
`

any idea?
(may-be related to this issue not really solvaed (the user found a workaround): https://github.com/billw2/rpi-clone/issues/59)

darrepac, 2019-08-12:
going back to 1.7 version make it works. Shame that there seems to have no support here...

framps, 2019-08-12:
> Shame that there seems to have no support here...

There are 153 forks of this repo and some may be maintained :wink:

tablatronix, 2020-12-03:
Same issue

blackjack75, 2021-11-11:
I had this issue tryint clone my RetroPie RPI4 (runs on dietpi) sd-card to an external SSD.  I read (somewhere?) rpi-clone miscalculated the sized of the destination partition and this caused the filesystem to go in read only mode. 

So I tried my luck. I created manually with fdisk

1) 256 MB partition (as vfat)
2) another partition of the rest of the dis (as ext4)

When I ran rpi-clone it didn't try to recreate the partitions so it worked beautifully, doing just the rsync. That's at least half the work I don't have to do manually.

BTW, this probably only happens with some disk structures. I just rpi-cloned a brand new install of raspberry OS  latest version from a RaspBerry 3 SD Card to an USB 2.0 HD and it worked without issues.