billw2/rpi-clone #79: RPI-CLONE to NAS

xCozmox, 2019-08-28:
Hello I have multiple Raspberries running in my network. 
I want to make a clone on weekly basis of each of them to my NAS. 
I mounted the NAS into my Raspberry /mnt/backup 

and now my question: Can I use rpi-clone to make a weekly backup into the mounted location /mnt/backup/image_ddmmyyyy.img ?

Thanks a lot for your help.

billw2, 2019-08-30:
rpi-clone can't backup to .img files, but I will look into adding that option.

SolarDuck, 2019-08-30:
Re backing up to an image file, there has been some initial work done in Issue#53. When coupled with SSHFS the file can be network connected. The use of a sparse file in Issue#53 may be worth looking into but may not be well understood by some users. While looking at use of an image file, how about also adding some compression options too, please? The other backup software I have been using for my RPi systems is fsarchvier, which adopts a different approach but has some nice features that rpi-clone could be enhanced with.

Thank you for all the great work.

xCozmox, 2019-09-02:
I found this. 
[bkup_rpimage](https://github.com/lzkelley/bkup_rpimage)

It has the .img option and it has the gzip option. Maybe you can use it for this project?
Greets, xcozmox

xCozmox, 2019-09-04:
I created a blog entry on the automation of a raspberry pi towards a NAS.
[pi-backup](https://cozmopics.de/2019/09/03/pi-backup/)
Maybe there is some useful input for you?