billw2/rpi-clone #83: Original SD card gets mounted

NickD9, 2019-10-23:
I removed my original SD card, inserted a new one flashed with Raspbian Lite and then rebuilt the system.... all good so far. It's been running for a few months so thought it was time to clone it, and wanted to use rpi-clone to clone it back to the original card I removed.
I put the original in a USB card reader, inserted it into the Pi, and it mounted the EXT4 partition as "/", and totally refused to unmount it. "/" was already occupied by the resident SD card, I had TWO root partitions, and nothing would resolve the issue until I rebooted and reformatted the original card using a Windows PC before reinserting into the Pi. All went fine after that, rpi-clone did its thing and I've got my cloned card.

Any ideas why the original got mounted, and will it happen again if pop the clone back in for an incremental update?

billw2, 2019-10-23:
On Wed, 23 Oct 2019 13:57:26 -0700
NickD9 <notifications@github.com> wrote:

> I removed my original SD card, inserted a new one flashed with Raspbian Lite and then rebuilt the system....
> all good so far. It's been running for a few months so thought it was time to clone it, [...]

I think this could happen if both your original and new SD cards had the same disk id (the PARTUUID)
and they would for sure if both were imaged from the same .img file.  Then because the PARTUUID is
referenced in fstab or maybe because auto mounting is enabled, the system automatically mounted the
SD card in the card reader on top of root.

rpi-clone does a PARTUUID check of the card reader disk and the booted disk. If the numbers are
the same, rpi-clone changes the destination PARTUUID so you should not have that problem again
when you do the next incremental clone.

Bill

NickD9, 2019-10-24:
Ah yes, they were both from the same image I’d previously taken, so you’re absolutely right. Would the OS mounting it be seen as a feature or an OS bug:-)
Many thanks for your help, rpi-clone is a work of genius and much appreciated. Any chance of it creating img files in the near future ;-)
Regards, Nick

billw2, 2019-10-24:
I think allowing this kind of mounting is for a feature having to with union or overlay files systems in
Linux. But I don't know much about them.  

Others have asked about backups to a .img file and I have thought I would see if it's something
I could add.  It's just that I need to find the time to look into it.