billw2/rpi-clone #85: Raspbian Buster

antoniovalenzuela, 2019-11-21:
Please help.
The copy does not end, it hangs.
Before cloned with Stretch without problems

 rpi-clone -f -p 256M sda

Booted disk: mmcblk0 16.1GB                Destination disk: sda 16.1GB
---------------------------------------------------------------------------
Part      Size    FS     Label           Part   Size  FS  Label
1 /boot   256.0M  fat32  --
2 root     14.8G  ext4   rootfs
---------------------------------------------------------------------------
== Initialize: IMAGE partition table - forced by option ==
1 /boot               (51.5M used)   : RESIZE  MKFS  SYNC to sda1
2 root                (2.7G used)    : RESIZE  MKFS  SYNC to sda2
---------------------------------------------------------------------------
-p 256M                : resize /boot to 524288 blocks of 512 Bytes.
Run setup script       : no.
Verbose mode           : no.
-----------------------:
** WARNING **          : All destination disk sda data will be overwritten!
-----------------------:

Initialize and clone to the destination disk sda?  (yes/no): yes
Optional destination ext type file system label (16 chars max):

Initializing
  Imaging past partition 1 start.

  => dd if=/dev/mmcblk0 of=/dev/sda bs=1M count=8 ...
  Resizing both destination disk partitions ...
    Resize success.
  Changing destination Disk ID ...
  => mkfs -t vfat -F 32  /dev/sda1 ...
  => mkfs -t ext4  /dev/sda2 ...

Syncing file systems (can take a long time)
Syncing mounted partitions:
  Mounting /dev/sda2 on /mnt/clone
  => rsync // /mnt/clone with-root-excludes ...

I had to kill the processes

antoniovalenzuela, 2019-12-02:
@billw2 Hi
News?

billw2, 2019-12-04:
I have not seen a hang like that.  Run in verbose mode:  rpi-clone -v ...
to see the last rsync output where it hangs.

antoniovalenzuela, 2019-12-04:
> I have not seen a hang like that. Run in verbose mode: rpi-clone -v ... to see the last rsync output where it hangs.
> […](#)

Ok I'll try

nomad2246, 2022-06-19:
> I have not seen a hang like that. Run in verbose mode: rpi-clone -v ... to see the last rsync output where it hangs.
> […](#)

Hi, I have the same problem. 
The copy doesn't quite finish, I think it's at the unmount point. Although at the beginning of the backup it is not set a new disk ID.
The app pause on:
"sent 63,514,294 bytes  received 55,216 bytes  1,842,594.49 bytes/sec
total size is 11,979,134,955  speedup is 188.44

"

I have the same problem with a virtualized debian 11 x64, and on an intel nuc x64 with debian 11.

Thanks a lot

"
javi@debiangui:~$ sudo sys-clone -v sdb
/usr/sbin/grub-install

Booted disk: sda 32,2GB                    Destination disk: sdb 150,5GB
---------------------------------------------------------------------------
Part     Size  FS  Label           Part   Size  FS    Label            
1 root    30   0G  ext4            --    1        30  0G     ext4  --  
---------------------------------------------------------------------------
== SYNC sda file systems to sdb ==
/                     (10,7G used)   : SYNC to sdb1 (30,0G size)
---------------------------------------------------------------------------
Run setup script       : no.
Run grub               : grub-install --root-directory=/mnt/clone /dev/sdb
Verbose mode           : yes.
-----------------------:

Ok to proceed with the clone?  (yes/no): yes
Destination disk has same Disk ID as source, changing it.
  Failed to set a new Disk ID.

Syncing file systems (can take a long time)
Syncing mounted partitions:
  Mounting /dev/sdb1 on /mnt/clone
  => rsync // /mnt/clone with-root-excludes ...sending incremental file list
dev/
etc/
etc/resolv.conf
..........
"

nomad2246, 2022-06-19:
More tests,
The program stops between lines 1760 and 1768. If I delete these lines of code the program continues.
I imagine it is because between lines 1633 and 1649 the program is not able to change the new Disk ID.

Sorry I can't be of more help, I have no idea of programming.

line: 1633-1649
"
line=$(fdisk -l /dev/$dst_disk | grep "Disk identifier:")
dst_disk_ID=${line#*x}
if [ "$dst_disk_ID" == "$src_disk_ID" ]
then
        qecho "Destination disk has same Disk ID as source, changing it."
        new_id=$(od -A n -t x -N 4 /dev/urandom | tr -d " ")
        qprintf "x\ni\n0x$new_id\nr\nw\nq\n" | fdisk /dev/$dst_disk | grep changed
        sync
        sleep 2
        partprobe "/dev/$dst_disk"
        sleep 2

        line=$(fdisk -l /dev/$dst_disk | grep "Disk identifier:")
        dst_disk_ID=${line#*x}
        if [ "$dst_disk_ID" == "$src_disk_ID" ]
        then
                qecho "  Failed to set a new Disk ID."
        fi
fi
"

line: 1760-1768
"
fi

if grep -q $src_disk_ID $fstab
then
        qecho "Editing $fstab PARTUUID to use $dst_disk_ID"
        sed -i "s/${src_disk_ID}/${dst_disk_ID}/g" "$fstab"
elif [ "$edit_fstab_name" != "" ] && grep -q ${src_part_base} $fstab
then
        qecho "Editing $fstab references from $src_part_base to $edit_fstab_name"
        sed -i "s/${src_part_base}/${edit_fstab_name}/" "$fstab"
fi
"

closed 2024-08-31 by antoniovalenzuela