billw2/rpi-clone #88: Not an issue - a Plea

scargill, 2020-02-11:
This could well beme missing something.. I use RPI-clone every day - essential stuff - sudo rpi-clone -U sda

After making a clone and putting that into the PI, I'm assuming that the clone is merely a copy of the original including timestamps. If so then if one makes several clones there is no way to see - once emplyed, when the clone was created.  SO, if an RPI SD is broken and you have a choice of several clonses to use, is there a way once the clone is put into the PI to see ehen THAT clone was created?

I kno - probaly a daft query....

lbussy, 2020-02-11:
Touch a file named something interesting right before you do the backup?  That way it will be the time / date it was taken.

scargill, 2020-02-11:
Thanks for that Lee - sometimes I miss the obvious...