billw2/rpi-clone #89: Don't know how to partition the destination disk!

Lorccan, 2020-02-24:
I'm using an RPI 3B+ booted from a 64GB USB stick using Raspbian GNU/Linux 9 (stretch)

I want to clone this to an identical USB stick, but when I do:

`sudo rpi-clone sdb`

I get:

`Boot and root are on different disks and it's not a SD card boot.
Don't know how to partition the destination disk!`

I am sure I had this working on a previous system with the same setup and - I'm pretty sure - the same OS version, but I am stuck.

I have tried various switches to force the clone, but none of these nor verbose logging gets me anywhere.

Any ideas please?

billw2, 2020-02-25:
On Mon, 24 Feb 2020 09:10:07 -0800
Lorccan <notifications@github.com> wrote:

> I'm using an RPI 3B+ booted from a 64GB USB stick using Raspbian GNU/Linux 9 (stretch)
>
> I want to clone this to an identical USB stick, but when I do:
>
> `sudo rpi-clone sdb`
>
> I get:
>
> `Boot and root [...]

rpi-clone uses the source disk to create a partition table on the destination disk but if you are booted
with root and /boot on different disks there is confusion on what to use as a source for the partition table.
So in this case rpi-clone punts and you have to make the partitions and filesystems on your destination
disk by hand to make sure you get what you need on the destination disk.

Look at example 7 on the https://github.com/billw2/rpi-clone web page for the commands you need to run
to do this.  That example makes a 3 partition destination disk but for your case it is likely you want just
two partitions.  Also make the first partition size 256M instead of 100M as shown to be compatible with
recent Raspbian releases.  Be sure that you create partitions and make the filesystems on the destination
disk (file systems will be made for you if you use gparted).

Bill

Lorccan, 2020-02-25:
Thanks Bill

I followed the example and setup new formatted partitions that matched the sizes on the source USB stick as closely as possible. Cloning would not proceed without the -F flag (I got the same error as above), but using it did allow it to proceed normally.

Curiously - or maybe not - once the forced clone had completed once, I was then able to do a non-forced clone afterwards. Is that what you'd expect?