billw2/rpi-clone #90: Feature request - any update on possible creation of an image output file?

CaptainMidnight, 2020-03-08:
Hi billw2

Reading through the various issues ( esp. issue #53 ) is there any update on the possibility of this functionality being added to rpi-clone?

Cureently us rpi-clone for card duplication and switch to 'clonepi' for compressed image creation requirements - it would be great if I could do this with just rpi-clone..........

framps, 2020-04-15:
Last code update of rpi-clone is more than 9 months ago. I have no idea why @billw2 suspended support for his very useful tool. You may check [raspiBackup](https://github.com/framps/raspiBackup) whether it fit's for you. It's still stupported.

p1r473, 2021-11-27:
@billw2 can you merge in the image creation from github.com/SeeSpotRun/rpi-clone?