billw2/rpi-clone #92: Fails to work on p4 buster 10

airscapes, 2020-05-11:
Not an expert but was working with a friend who has the same setup which works.. mine does not.. 
pi@kodipi2:~$ sudo  rpi-clone -f -p 256M sda

  Destination disk partition /dev/sda1 is mounted on /media/pi/KODIPI2.
  The clone cannot proceed unless it is unmounted.
Do you want to unmount /media/pi/KODIPI2?  (yes/no): yes

Booted disk: mmcblk0 63.9GB                Destination disk: sda 63.9GB
---------------------------------------------------------------------------
Part                   Size    FS     Label           Part   Size    FS     Label
1 /media/pi/RECOVERY    63.0M  fat16  --              1       59.5G  fat32  --
2                       59.4G  EXT    --                                      
                                                                              
                                                                              
5 /media/pi/SETTINGS    31.5M  ext4   SETTINGS                                
6 /boot                255.5M  fat32  --                                      
7 root                  59.1G  ext4   root                                    
---------------------------------------------------------------------------
== Initialize: IMAGE partition table - forced by option ==
1 /media/pi/RECOVERY  (38.5M used)   : RESIZE  MKFS  SYNC to sda1
5 /media/pi/SETTINGS  (1.5M used)    : MKFS  SYNC to sda5
6 /boot               (52.0M used)   : MKFS  SYNC to sda6
7 root                (6.9G used)    : RESIZE  MKFS  SYNC to sda7
---------------------------------------------------------------------------
-p 256M                : resize /boot to 524288 blocks of 512 Bytes.
Run setup script       : no.
Verbose mode           : no.
-----------------------:
** WARNING **          : All destination disk sda data will be overwritten!
-----------------------:

Initialize and clone to the destination disk sda?  (yes/no): y
Optional destination ext type file system label (16 chars max): 

Initializing
  Imaging past partition 1 start.
  => dd if=/dev/mmcblk0 of=/dev/sda bs=1M count=8 ...
/usr/local/sbin/rpi-clone: line 1432: 129024
65534
524286: syntax error in expression (error token is "65534
524286")
Destination disk has same Disk ID as source, changing it.
Disk identifier changed from 0x1bb3f472 to 0x64735e71.
Error: Invalid partition table on /dev/sda -- wrong signature 0.

Syncing file systems (can take a long time)
Syncing mounted partitions:
  Mounting /dev/sda7 on /mnt/clone
mount: /mnt/clone: special device /dev/sda7 does not exist.
    Mount failure of /dev/sda7 on /mnt/clone.
Aborting!

airscapes, 2020-05-11:
Removed partition and recreated single fat32 partition and formated, then reran with -x see attached
[error.out.txt](https://github.com/billw2/rpi-clone/files/4612287/error.out.txt)

airscapes, 2020-05-11:
This is the source card
Disk /dev/mmcblk0: 59.5 GiB, 63864569856 bytes, 124735488 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x1bb3f472

Device         Boot  Start       End   Sectors  Size Id Type
/dev/mmcblk0p1        8192    137215    129024   63M  e W95 FAT16 (LBA)
/dev/mmcblk0p2      137216 124735487 124598272 59.4G  5 Extended
/dev/mmcblk0p5      139264    204797     65534   32M 83 Linux
/dev/mmcblk0p6      204800    729085    524286  256M  c W95 FAT32 (LBA)
/dev/mmcblk0p7      729088 124735487 124006400 59.1G 83 Linux

airscapes, 2020-05-11:
source disk parted output
Model: SD SN64G (sd/mmc)
Disk /dev/mmcblk0: 63.9GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags: 

Number  Start   End     Size    Type      File system  Flags
 1      4194kB  70.3MB  66.1MB  primary   fat16        lba
 2      70.3MB  63.9GB  63.8GB  extended
 5      71.3MB  105MB   33.6MB  logical   ext4
 6      105MB   373MB   268MB   logical   fat32        lba
 7      373MB   63.9GB  63.5GB  logical   ext4

billw2, 2020-05-12:
On Mon, 11 May 2020 16:03:31 -0700
airscapes <notifications@github.com> wrote:
>
> Not an expert but was working with a friend who has the same setup which works.. mine does not.. 
> pi@kodipi2:~$ sudo  rpi-clone -f -p 256M sda

...

> source disk parted output
> Model: SD SN64G (sd/mmc)
> Disk /dev/mmcblk0: 63.9GB
> Sector size (logical/physical): 512B/512B
> Partition Table: msdos
> Disk Flags:
>
> Number  Start   End     Size    Type [...]

If this is NOOBS or PINN, then the /boot partition is in the middle of the partition mix and
a limitation of rpi-clone is that it cannot resize such a partition.  The -p 256M option is intended for
the Raspbian 2 partition image distribution where the /boot partition is partition 1.  Did you
start with an image containing Buster?  If so it should have started with a /boot at 256M.

So rpi-clone is getting confused and I should add a test for when this happens.

All you can do is try running rpi-clone without using the "-p 256M" option.  With that I hope
it will work better.  And unfortunately that's all it is, a hope, because I wrote the code with the
idea that it might work with NOOBS or PINN, but I do not run those distributions so have never
been able to test that.  Sorry about that.

Bill

airscapes, 2020-05-12:
When run without the -p option it still trys to resize the last partition. Not sure why it thinks it is in use, file system on the target was unmounted before rerunning. The OS was installed a few months ago and Yes I think I used NOOBS.  More or less followed the install instrutions I found with google.
pi@kodipi2:~$ cat /tmp/rpi-clone-output
Checking that no-one is using this disk right now ... FAILED

This disk is currently in use - repartitioning is probably a bad idea.
Umount all file systems, and swapoff all swap partitions on this disk.
Use the --no-reread flag to suppress this check.

Disk /dev/sdb: 59.5 GiB, 63864569856 bytes, 124735488 sectors
Disk model: STORAGE DEVICE  
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x1bb3f472

Old situation:

Device     Boot  Start       End   Sectors  Size Id Type
/dev/sdb1         8192    137215    129024   63M  e W95 FAT16 (LBA)
/dev/sdb2       137216 124735487 124598272 59.4G  5 Extended

>>> Script header accepted.
>>> Script header accepted.
>>> Script header accepted.
>>> Script header accepted.
>>> Created a new DOS disklabel with disk identifier 0x1bb3f472.
/dev/sdb1: Created a new partition 1 of type 'W95 FAT16 (LBA)' and of size 63 MiB.
Partition #1 contains a vfat signature.
/dev/sdb2: Created a new partition 2 of type 'Extended' and of size 59.4 GiB.
/dev/sdb3: Created a new partition 5 of type 'Linux' and of size 32 MiB.
/dev/sdb6: Start sector 204800 out of range.
Failed to add #6 partition: Numerical result out of range
Leaving.

====rpi-clone
==orig:
label: dos
label-id: 0x1bb3f472
device: /dev/mmcblk0
unit: sectors

/dev/mmcblk0p1 : start=        8192, size=      129024, type=e
/dev/mmcblk0p2 : start=      137216, size=   124598272, type=5
/dev/mmcblk0p5 : start=      139264, size=       65534, type=83
/dev/mmcblk0p6 : start=      204800, size=      524286, type=c
/dev/mmcblk0p7 : start=      729088, size=   124006400, type=83

==edited:
label: dos
label-id: 0x1bb3f472
device: /dev/mmcblk0
unit: sectors

/dev/mmcblk0p1 : start=        8192, size=      129024, type=e
/dev/mmcblk0p2 : start=      137216,  type=5
/dev/mmcblk0p5 : start=      139264, size=       65534, type=83
/dev/mmcblk0p6 : start=      204800, size=      524286, type=c
/dev/mmcblk0p7 : start=      729088,  type=83

airscapes, 2020-05-12:
I reinstalled full raspbian using the raspberry pi image tool rather than NOOBS and I now have a nice clean 2 partition, standard looking configuration.  Your program  ran perfectly and I will try booting from the card in the morning. Will close this after that is complete .

airscapes, 2020-05-13:
Booted right up.. after some searching it looks like the cause of the failure was not NOOBS but NOOBS lite.  My coworker who pointed me to this script had no issues and his system was installed with NOOBS but I downloaded NOOBSlite which created the very odd partition table.  Thanks and sorry for all the fuss.

closed 2020-05-13 by airscapes