billw2/rpi-clone #93: Volumio cloning fails

valiant66, 2020-05-15:
Hi.

I'm trying to install a touchscreen with Volumio, and have failed many many times.

So I decided to create backups of my Volumio install, sort of like save points in a game.

This is what I did: I installed Volumio on two identical 32GB Sandisk Ultra microSD cards and confirmed they would both boot the RPi.

I then started updating one of them, including installing rpi-clone.

At a certain time it was time for a backup, and I tried using rpi-clone to clone the boot card to the backup card.

Utter failure. The resulting card would not boot.

I used the -v option, and this is a bit of the result:

`sent 1,736,445,592 bytes  received 1,291,261 bytes  8,497,490.72 bytes/sec
total size is 1,834,211,563  speedup is 1.06

  Mounting /dev/sda1 on /mnt/clone/boot
  => rsync /boot/ /mnt/clone/boot  ...sending incremental file list
rsync: chown "/mnt/clone/boot/." failed: Operation not permitted (1)
./
rsync: chown "/mnt/clone/boot/.firmware_revision" failed: Operation not permitted (1)
rsync: chown "/mnt/clone/boot/.firmware_revision_kernel" failed: Operation not permitted (1)
rsync: chown "/mnt/clone/boot/bcm2708-rpi-b-plus.dtb" failed: Operation not permitted (1)
rsync: chown "/mnt/clone/boot/bcm2708-rpi-b.dtb" failed: Operation not permitted (1)

etc.
etc.
etc.

rsync: chown "/mnt/clone/boot/overlays/wittypi.dtbo" failed: Operation not permitted (1)

sent 5,087 bytes  received 22,162 bytes  18,166.00 bytes/sec
total size is 60,978,153  speedup is 2,237.81
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1183) [sender=3.1.1]

===============================
Done with clone to /dev/sda
   Start - 00:39:17    End - 00:42:48    Elapsed Time - 3:31

Cloned partitions are mounted on /mnt/clone for inspection or customizing. 

Hit Enter when ready to unmount the /dev/sda partitions ...     
  unmounting /mnt/clone/boot
  unmounting /mnt/clone
===============================
`
When I took the microSD card out of the USB holder and tried to boot the RPi from it... crickets.

So I thought I would try the -f flag, and force initialization of the microSD card first.

`volumio@volumio:~$ sudo rpi-clone sda -vf
[sudo] password for volumio: 
/usr/local/sbin/rpi-clone: line 571:  -1: substring expression < 0
Bad arg: -vf
Run rpi-clone with -h or no args for usage.
`
This confused me, since rpi-clone does not appear to have a -h flag.

So I tried the non-verbose version.

`volumio@volumio:~$ sudo rpi-clone -f sda
/usr/local/sbin/rpi-clone: line 571:  -1: substring expression < 0

Booted disk: mmcblk0 31.9GB                Destination disk: sda 31.9GB
---------------------------------------------------------------------------
Part         Size    FS     Label           Part   Size    FS     Label
1 /boot       61.0M  fat32  --              1       61.0M  fat32  --
2 /imgpart     2.3G  ext4   --              2        2.3G  ext4   volumio
3             27.4G  ext4   volumio_data    3       27.4G  ext4   volumio_data
---------------------------------------------------------------------------
== Initialize: IMAGE partition table - forced by option ==
1 /boot               (58.0M used)   : MKFS  SYNC to sda1
2 /imgpart                           : MKFS  SYNC to sda2
3                     (753.0M used)  : RESIZE  MKFS  SYNC to sda3
---------------------------------------------------------------------------
Run setup script       : no.
Verbose mode           : no.
-----------------------:
** WARNING **          : All destination disk sda data will be overwritten!
-----------------------:

Initialize and clone to the destination disk sda?  (yes/no): y
Optional destination ext type file system label (16 chars max): 

Initializing
  Imaging past partition 1 start.
  => dd if=/dev/mmcblk0 of=/dev/sda bs=1M count=4 ...
  Resizing destination disk last partition ...
    Resize failed.  See /tmp/rpi-clone-output.
    Try running rpi-clone again.

Re-reading the partition table failed.: Device or resource busy
`
I tried manually unmounting the microSD card using umount then running the same command, and it made no difference.

So. I've tried this with two different USB adapters and three different microSD cards. I've reformatted and reflashed the cards and they still work, so the cards are not corrupted.

My goal is to be able to make bootable backups so that when I screw things up trying to get the touchscreen to work (the weeks-long failure of getting the drivers to work is why I decided to try making bootable backcups) I can step back easily and try again.

Any ideas of what I'm doing wrong and how to fix it?

Thanks!

billw2, 2020-05-15:
On Thu, 14 May 2020 19:31:07 -0700
valiant66 <notifications@github.com> wrote:

>   => rsync /boot/ /mnt/clone/boot  ...sending incremental file list
> rsync: chown "/mnt/clone/boot/." failed: Operation not permitted (1)
> ./
> rsync: chown "/mnt/clone/boot/.firmware_revision" failed: Operation not permitted (1)

...

>   Imaging past partition 1 start.
>   => dd if=/dev/mmcblk0 of=/dev/sda bs=1M count=4 ...
>   Resizing destination disk last partition ...
>     Resize failed.  See /tmp/rpi-clone-output.

It looks like multiple programs are failing to write to the SD card for some reason
and it's unlikely that rpi-clone can then succeed with a clone.

With /dev/sda1 mounted either still on /mnt/clone/boot or by hand mounted somewhere,
can you write a file to the SD card?  Then unmount the partition and remount it and see if
the file really did get saved to the SD card?

Also, what's in the /tmp/rpi-clone-output file?

valiant66, 2020-05-15:
I created a test directory in /media/boot and copied a README.md file to it.

I then unmounted the volumes and physicall moved the microSD card to a different USB port.

I then checked and the directory and file were both where I put them. I opened the file just to make sure it was OK, and it was as expected.

This is the content of the rpi-clone-output file:

No it's not. I can't figure out how to copy from nano and paste into chromium in debian. It appears to lack a universal copy function.

valiant66, 2020-05-15:
I thought I could copy and paste from Terminal in Mac OS, but that doesn't work either.

So instead here's a screengrab of the rpi-clone-output file:

[image]

valiant66, 2020-05-15:
Ah-ha!

I can save the output from the terminal session to a text file and use that. So here's the above in text format, if that helps:

                       ___                                      
                      /\_ \                        __           
         __  __    ___\//\ \    __  __    ___ ___ /\_\    ___   
        /\ \/\ \  / __`\\ \ \  /\ \/\ \ /' __` __`\/\ \  / __`\ 
        \ \ \_/ |/\ \L\ \\_\ \_\ \ \_\ \/\ \/\ \/\ \ \ \/\ \L\ \
         \ \___/ \ \____//\____\\ \____/\ \_\ \_\ \_\ \_\ \____/
          \/__/   \/___/ \/____/ \/___/  \/_/\/_/\/_/\/_/\/___/ 
        
             Free Audiophile Linux Music Player - Version 2.0

          C 2015 Michelangelo Guarise - Volumio Team - Volumio.org
                               

Volumio Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
volumio@volumio:~$ sudo nano /tmp/rpi-clone-output 
[sudo] password for volumio: 

  GNU nano 2.2.6                      File: /tmp/rpi-clone-output                                                    

sfdisk: Checking that no-one is using this disk right now ...
sfdisk: BLKRRPART: Device or resource busy
sfdisk:
This disk is currently in use - repartitioning is probably a bad idea.
Umount all file systems, and swapoff all swap partitions on this disk.
Use the --no-reread flag to suppress this check.
sfdisk: Warning: partition 1 does not end at a cylinder boundary
sfdisk: Warning: partition 2 does not start at a cylinder boundary
sfdisk: Warning: partition 2 does not end at a cylinder boundary
sfdisk: Warning: partition 3 does not start at a cylinder boundary
sfdisk: Warning: partition 3 does not end at a cylinder boundary

Disk /dev/sda: 30436 cylinders, 64 heads, 32 sectors/track
Old situation:
Units: cylinders of 1048576 bytes, blocks of 1024 bytes, counting from 0

   Device Boot Start     End   #cyls    #blocks   Id  System
/dev/sda1   *      0+     61-     62-     62500    c  W95 FAT32 (LBA)
/dev/sda2         61+   2384-   2324-   2378906   83  Linux
/dev/sda3       2384+  30436-  28053-  28725569+  83  Linux
/dev/sda4          0       -       0          0    0  Empty
New situation:
Units: sectors of 512 bytes, counting from 0

   Device Boot    Start       End   #sectors  Id  System
/dev/sda1   *         1    125000     125000   c  W95 FAT32 (LBA)
/dev/sda2        125001   4882812    4757812  83  Linux
/dev/sda3       4882813  62333951   57451139  83  Linux
/dev/sda4             0         -          0   0  Empty
Successfully wrote the new partition table

sfdisk: BLKRRPART: Device or resource busy
sfdisk: The command to re-read the partition table failed.
Run partprobe(8), kpartx(8) or reboot your system now,
before using mkfs
sfdisk: If you created or changed a DOS partition, /dev/foo7, say, then use dd(1)
to zero the first 512 bytes:  dd if=/dev/zero of=/dev/foo7 bs=512 count=1
(See fdisk(8).)
Re-reading the partition table ...

====rpi-clone
==orig:
# partition table of /dev/mmcblk0
unit: sectors

/dev/mmcblk0p1 : start=        1, size=   125000, Id= c, bootable
/dev/mmcblk0p2 : start=   125001, size=  4757812, Id=83
/dev/mmcblk0p3 : start=  4882813, size= 57451139, Id=83
/dev/mmcblk0p4 : start=        0, size=        0, Id= 0

==edited:
# partition table of /dev/mmcblk0
unit: sectors

/dev/mmcblk0p1 : start=        1, size=   125000, Id= c, bootable
/dev/mmcblk0p2 : start=   125001, size=  4757812, Id=83
/dev/mmcblk0p3 : start=  4882813,  Id=83
/dev/mmcblk0p4 : start=        0, size=        0, Id= 0

^G Get Help        ^O WriteOut        ^R Read File       ^Y Prev Page       ^K Cut Text        ^C Cur Pos
^X Exit            ^J Justify         ^W Where Is        ^V Next Page       ^U UnCut Text      ^T To Spell