billw2/rpi-clone #97: Feature request: Change of IPaddress

SargonofAssyria, 2020-05-28:
It would be nice if the rpi-clone-setup script could change the IPaddress of the clone too.

billw2, 2020-06-19:
There are too many variations on neworking setups for rpi-clone to do that reliably.  It's what the rpi-clone-setup script is for.  Edit that to change your IP in hosts files and networking files. Then run rpi-clone with the -s option option so it runs the setup script.  I do this for my clones where I clone to several different Pi installations with hostnames ranging like rpi0, rpi1, etc.  So I run rpi-clone on rpi0 and run a setup for rpi1 with:
   rpi-clone -s rpi1

Then my rpi-clone-setup change files for hostame  (/etc/hostname) and for networking.  All my Pis connect via dhcp but my  setup script sets up a fixed LAN IP address for each one by setting the IP address with a "ip address add ..." command in a /etc/dhcpcd.exit-hook script.  Your setup is likely different and I would not know what to tell rpi-clone to do.

SargonofAssyria, 2020-06-20:
Ok, tnx. I will do by scripting.

closed 2020-06-20 by SargonofAssyria