billw2/rpi-clone #98: RPIVLONE and the new USB BOOT on Pi4 capability

scargill, 2020-05-31:
Pi4 - I have successully made an SD of the new Raspbian that works both as SD and USB for USB boot and I've used rpi-clone to make a copy to a real USB which also boots.. I need to get back my working setup of a slightly older SD which will NOT USB boot. Can I somehow use pi-clone ONLY to overwite the second partition to make a new USB boot compatibe version of my workingg installation? Any ideas?

scargill, 2020-05-31:
Also, with my USB running the show I tried making a copy using rpiclone to a usb3 HD . The rpi-clone response was  

sudo rpi-clone -f -U sdb
Warning: Not all of the space available to /dev/sdb appears to be used, you can fix the GPT to
use all of the space (an extra 95230128 blocks) or continue with the current setting?
parted: invalid token: ?

With no instruction as to how to "fix the GPT" ? Any thoughts (not rude) - I've no idea what a GPT is or what to do next but I want to use the whole drive.....

billw2, 2020-06-19:
For SD card to USB boot there are several ways to go so I'll show some examples that maybe will cover your cases:

You have an SD card booting on the Pi 4 and want to make it USB bootable:
1) For safety, make a pure SD backup.  With an SD card in a USB card reader at sdb:
    rpi-clone sdb
2) Now make your working SD card boot to a USB disk at sda.  Clone to the sda with -l option:
   rpi-clone -l sda

Reboot and the SD card will transfer the booting to the USB disk. After boot, the SD card is not in use and the root will be on sda2 and /boot will be sda1

3a) With the USB disk booted, clone only the root back to the SD card still in mmcblk0:
    rpi-clone -m / mmcblk0
3b) or better, especially if you have apt upgraded while running of USB, clone both root and /boot back to the SD card, but preserve the SD card to USB boot setup (again use -l option):
   rpi-clone -l mmcblk0

4) If you upgrade and rpi-eeprom version has USB booting built in, do a rpi-clone as in 3b) so the SD card will have the new eeprom files.  Use the -l option!!!  Now reboot letting the SD card do the boot to USB.  This should update the eeprom so you no longer need the  SD card to boot.  So, pull it out and reboot.

5) With native USB booting working, you no longer need to use the -l option when cloning from SD card to USB or USB to SD.

The GPT error is from the parted command and I would need more debug info to know why you get that.  Is the USB 3 HD huge?  Maybe a partition sizing problem.  The -f copies the partition table and that could be incompatible.  In that case you would need to manually partition the USB 3 HD and not use the -f option.  See the examples for manually partitioning.