billw2/rpi-clone #105: noobie help on what I did wrong :-)

mahjohn, 2020-09-24:
Attempting to clone my 64GB card to my 128GB card, then vice versa in a couple of weeks after any substantial changes. 128GB SD card is in a USB card reader.

Using Example #1 i ran `sudo rpi-clone sdb` which produced the image below and looks to be correct.

[image]

I select 'Y' and it runs to completion. However, once I swap the cards to test out the new clone copy, it doesn't boot. What have I done wrong? 
Thanks!

[image]

jcbvm, 2020-10-22:
I have the same issue... also newby here

gatman1104, 2020-11-26:
Same here.  Just tried to clone from SD card to a USB 3 thumb drive.  Everything worked fine with no errors, but USB drive will not boot (this Pi has been set to boot from USB and has successfully done so with a thumb drive created with Etcher).  Please help.

nishantbb, 2020-12-06:
I have the same issue. Additionally, I cannot boot from the original SD card now. Is there guidance on reverting any changes to the source during this process?