billw2/rpi-clone #106: USB Boot and USB Clone - Indeterminate Boot Device - Use SED on cmdline.txt?

eddiem5, 2020-10-05:
Hello,

Love rpi-clone - Thank you..

I have a PI doing USB boot and another USB drive for a clone backup.

The problem is when the system is rebooted, it is indeterminate which device gets the boot...

I was hoping the -l flag from rpi-clone would do the trick but it only works for SD cards...

Any ideas or help would be great..

robustini, 2021-01-12:
Same problem here, it is due to the bootloader being able to start from USB.
I advise you to use version 1.71, never wrong, this new V2 at least here has only problems.

robustini, 2021-01-13:
Unzip to /usr/local/sbin, then "sudo chmod +x rpi-clone" and use it, end of problems.

[rpi-clone.zip](https://github.com/billw2/rpi-clone/files/5807530/rpi-clone.zip)

framps, 2021-01-13:
@robustini That's a very old version 1.7.1 of rpi-clone. Current version is 2.0.22. Would be great if you create an PR on 2.0.22.

robustini, 2021-01-14:
> 
> 
> @robustini That's a very old version 1.7.1 of rpi-clone. Current version is 2.0.22. Would be great if you create an PR on 2.0.22.

I know it's old, but at least this one almost always works!

framps, 2021-01-14:
> I know it's old, but at least this one almost always works!
> The result with V2.0.22:

It's great you provided a workaround for this issue :+1: I thought you created some fix in 1.7.1 and will be able to roll in this fix into the current version.