billw2/rpi-clone #110: Feature request: Mount after clone

folkeorg, 2020-12-06:
It seems that the SD-card is not mounted after cloning. I would like a switch e.g. "-M" (can't find it in the switch-list) to have the card mounted and readable files afterwards. Now I need to mount it manually or remove/insert the card to get it mounted.

I guess the program have all the mount parameters stored so a mount should be easy at the end of the cloning.

Apart from that the program is working great! Now I always have a ready-to-swap-card if the booted one fails (which it will eventually).

framps, 2020-12-06:
There is no additional option required in rpi-clone :wink: 

Why don't you just create an alias which executes rpi-clone followed by a mount command? Or you create a small script called my_rpi-clone which calls rpi-clone followed by a mount?

folkeorg, 2020-12-08:
Well, first of all I'm lazy. I like all-in-one-solutions. Secondly I'm really not that skilled in xNIX to be able to mount the SD-card manually and correctly.. I guess this last part is the real reason..