billw2/rpi-clone #113: Operation not permitted...

fmarzocca, 2021-01-25:
I don't understand the meaning of these errors....

```
Syncing file systems (can take a long time)
Syncing mounted partitions:
  Mounting /dev/sda2 on /mnt/clone
  => rsync // /mnt/clone with-root-excludes ...
  Mounting /dev/sda1 on /mnt/clone/boot
  => rsync /boot/ /mnt/clone/boot  ...rsync: chown "/mnt/clone/boot/." failed: Operation not permitted (1)
rsync: chown "/mnt/clone/boot/COPYING.linux" failed: Operation not permitted (1)
rsync: chown "/mnt/clone/boot/LICENCE.broadcom" failed: Operation not permitted (1)
rsync: chown "/mnt/clone/boot/bcm2708-rpi-b-plus.dtb" failed: Operation not permitted (1)
.....
.....
(lot of lines)
```

framps, 2021-01-25:
You either don't have the required access rights or the filesystem you use does not support Linux filesystem attributes. Don't use NTFS or any fat filesystem. Just use EXT3/4 or btrfs.

fmarzocca, 2021-01-25:
Raspberry's boot partition is always vfat:

```
/dev/mmcblk0p1: LABEL_FATBOOT="boot" LABEL="boot" UUID="F661-303B" TYPE="vfat" PARTUUID="b5a15f89-01"
/dev/mmcblk0p2: LABEL="rootfs" UUID="8d008fde-f12a-47f7-8519-197ea707d3d4" TYPE="ext4" PARTUUID="b5a15f89-02
```"
so, when you first "force" imaging on a new SD, also that partition will be vfat.

bonzer2u, 2021-02-01:
rpi-clone is totally dead for me now. rpi-4b 4gb, fully updated. I hope it gets addressed in the near future.
However, sd-card-copier is still functional as a backup.
Tried to clone a 32gb used of a 128gb usb flash drive to both a 64gb and 128gb sd-card with the following errors, attached.
[rpi-clone-error-evo64.txt](https://github.com/billw2/rpi-clone/files/5905450/rpi-clone-error-evo64.txt)
[rpi-clone-error-san128.txt](https://github.com/billw2/rpi-clone/files/5905451/rpi-clone-error-san128.txt)

sob, 2021-03-01:
Something is broken here with the /boot partition as I got the same results. Workaround was to dd the boot partition directly and then edit the cmdline.txt (something like this if I recall - it's not showing up in my bash_history):
`dd if=/dev/mmcblkp01 of=/dev/sda1 bs=10M count=30`

and then change the root partition in /dev/sda1->cmdline.txt (mine was corrupted as well somehow)

hendrix04, 2022-10-12:
Anyone figure this out? I am currently running into the same issue on my RPi 4.