billw2/rpi-clone #114: Feature request : clone to img file

stevenmache, 2021-04-11:
Hi, many thanks for this script, I use it as my main backup utility ^^

I would like to have backup in img file, runned from crom. I am using dd but with 2 major problems:
* it takes 3-8 hours (instead of 12 min with rpi-clone)
* my backup is the size of the sd card so 60Go instead of 13Go with rpi-clone

Well, if I could use rpi-clone in a cron to make a img file it would be the best backup tool !

For use with cron, it would need to use disk uuid instead of sdN (to avoid dramatic bad luck) 
For img file, maybe it is already possible with /dev/loop virtual block device ? But I haven't tried. 

Many thanks
Steven

PotemkinMcNasty, 2021-04-26:
+1 for this feature

fjpdevries, 2021-04-26:
Ik ben tot en met dinsdag 27 april 2021 afwezig en zal mijn email niet of beperkt kunnen lezen en beantwoorden. 

Met vriendelijke groet,
FSE Turnstiles b.v.
Jos de Vries

paloj, 2021-06-02:
+1 for this request

karuniko, 2021-07-27:
+1 for this feeature

framps, 2021-07-30:
Not sure whether this is an alternative for you folks: I wrote a script [raspiBackup](https://github.com/framps/raspiBackup) which also can create backups with rsync. A helper script[ raspiBackupRestore2Image.sh](https://github.com/framps/raspiBackup)  restores the created rsync backup into a dd image and compresses the image to it's minimum size with pishrink.

AzureRaptor, 2023-03-11:
> Not sure whether this is an alternative for you folks: I wrote a script [raspiBackup](https://github.com/framps/raspiBackup) which also can create backups with rsync. A helper script[ [...]

I know it's been a while on this thread but I just wanted to thank you _very_ much for this.  I've been trying various methods to do this all "by hand" and it's all been relatively kludgy until this.  _Thank you._

framps, 2023-03-11:
@AzureRaptor Glad raspiBackup helps you to manage your backups. Do you know there is a [beta](https://github.com/framps/raspiBackup/issues/553#issuecomment-1328109659) out there which supports Ubuntu and will be included in next release of raspiBackup?

AzureRaptor, 2023-03-11:
Even better news!  Since rpi-clone (99%, it misses cmdline.txt) works for Ubuntu it hadn't even occurred to me to check on that.  Ubuntu is indeed my preferred OS on the Pi 4B.  Recently upgraded to Jammy and it's been...interesting.