billw2/rpi-clone #121: Booting with initramfs on Ubuntu 18.04.2.LTS

Nixies, 2021-06-06:
Hello,

I'm trying to clone an sd card on which is installed an Ubuntu 18.04.2 LTS. During the cloning process everything goes ok but when I try the cloned sd card into the rpi  3b+ I land into initramfs prompt.
When I type exit it gives me the following message:

Gave up wainting for root file system device. Common problems:
- Boot args (cat /proc/cmdline)
   - Check rootdelay= (did the system wait long enough)
- Missing modules (cat /proc/modules; ls /dev)
ALERT! LABEL=writable does not exist. Dropping to a shell!

I tried to change the delay value without success.
The original card work perfectly on the rpi 3b+.

Do you have any idea ?
Thank you very much.

Nixies, 2021-06-13:
Hello,
Finally I solved the problem:
1- add Label to the root partition
2- modify the fstab: replace the LABEL="writable" by the PARTUUID value of this partition.

closed 2021-06-13 by Nixies

SomebodySysop, 2021-12-12:
> Hello, Finally I solved the problem: 1- add Label to the root partition 2- modify the fstab: replace the LABEL="writable" by the PARTUUID value of this partition.

I just cloned a Raspberry Pi 4 system using example 1), and but it into new Raspberry Pi 4 system and got the same exact error messages. This is kind of new to me.   Could you elaborate just a bit on how you did steps 1 and 2.

How did you add Label to the root partition?

As for the 2nd step, could you just show your fstab so we could see the change you made?

Thanks!

SomebodySysop, 2021-12-12:
I actually resolved the problem by using this to clone the sdcard: https://github.com/nwright-mcc/piclone_cmd

New system booted immediately.