billw2/rpi-clone #122: RPi4 Debian clone wont boot

diagonali, 2021-06-25:
I've successfully cloned an installation of Debian from an external USB drive to another USB drive, also tried cloning to an SD card. Booting from cloned drives/sd cards is unsuccessful.

When I remove the original bootable drive and reboot, the system wont boot from the cloned drive. This procedure works perfectly when I install Raspberry Pi OS but I have to use Debian for this install.

I notice mention of FSTAB and UUIDs in the documentation but despite researching what they are and how to make sure they're correctly setup I can't seem to figure it out. - Can someone help me figure out why my clone of a Debian install on a RPi4 won't boot? I just hangs somewhere around mention of bluetooth but I can't see how that would be related.

Is there anything obvious I need to do or check to make the cloned drive bootable?

Any help would be hugely appreciated.

Thanks

framps, 2021-06-25:
As far as I understand rpi-clone either supports partuuids or device names in /etc/fstab and /boot/cmdline.txt. If labels or uuids are used on your system rpi-clone will note be able to update them and your system will not boot.

You have to update them manually or use a backup tool which supports also uuids and labels.

diagonali, 2021-06-25:
> As far as I understand rpi-clone either supports partuuids or device names in /etc/fstab and /boot/cmdline.txt. If labels or uuids are used on your system rpi-clone will note be able to update them and your system will not boot.
> 
> You have to update them manually or use a backup tool which supports also uuids and labels.
> 
> Your issue seems to be a duplicate of #100

Thanks for reply and link to other issue already flagged. Does seem to explain the issue. It's unfortunately beyond my ability (I've googled for a guide but can't figure it out) currently to check the labels on the source drive and edit them on the destination to match so hopefully it will be added if there are future updates.

framps, 2021-06-25:
Would you please provide your /etc/fstab and /boot/cmdline.txt so I can check whether I'm right?

diagonali, 2021-06-25:
This is from my Debian Rpi4 system which boots from USB external drive:

fstab:
`# The root file system has fs_passno=1 as per fstab(5) for automatic fsck.
LABEL=RASPIROOT / ext4 rw 0 1
# All other file systems have fs_passno=2 as per fstab(5) for automatic fsck.
LABEL=RASPIFIRM /boot/firmware vfat rw 0 2`

I don't appear to have anything in my /boot/cmdline.txt

diagonali, 2021-06-25:
@framps I have seen this: [https://github.com/matthijskooijman/rpi-clone/blob/3441dace223580ca45d5877de0b8ccb952873747/rpi-clone](url) which may work. Just trying it now.

framps, 2021-06-25:
> LABEL=RASPIROOT / ext4 rw 0 1

Prooves you use LABELs on your system

> I don't appear to have anything in my /boot/cmdline.txt

I never have seen a Raspi without /boot/cmdline.txt. Frankly I have never seen a Debian on Raspi.

> @framps I have seen this: [matthijskooijman@3441dac](https://github.com/matthijskooijman/rpi-clone/commit/3441dace223580ca45d5877de0b8ccb952873747) which may work. Just trying it now.

I had a quick look into the code and it looks promising :+1: . Just give it a try and keep us posted :wink:

diagonali, 2021-06-25:
> > LABEL=RASPIROOT / ext4 rw 0 1
> 
> Prooves you use LABELs on your system
> 
> > I don't appear to have anything in my /boot/cmdline.txt
> 
> I never have seen a Raspi without /boot/cmdline.txt. Frankly I have never seen a Debian on Raspi.
> 
> > @framps I have seen this: [matthijskooijman@3441dac](https://github.com/matthijskooijman/rpi-clone/commit/3441dace223580ca45d5877de0b8ccb952873747) which may work. Just trying it now.
> 
> I had a quick look into the code and it looks promising 👍 . Just give it a try and keep us posted 😉

Didn't work, will need to try a full new copy will post back later. Thanks for your help! I'm running Debian on the Rpi so that I can run Home Assistant "officially".

buldezir, 2021-06-28:
@diagonali  had same problem cloning ubuntu server from bootable usb to sd card.
so the solution is to clone mentioned repo AND switch branch, the fix is not in master!

```
git clone https://github.com/matthijskooijman/rpi-clone.git
cd rpi-clone
git checkout keep-fs-labels
sudo ./rpi-clone mmcblk0
```

diagonali, 2021-06-29:
> @diagonali had same problem cloning ubuntu server from bootable usb to sd card.
> so the solution is to clone mentioned repo AND switch branch, the fix is not in master!
>
> ```
> git clone [...]

@buldezir Thanks for getting back to me with suggestion. I've tried it exactly as you describe using the exact commands you gave and unfortunately my Pi4 still won't boot. It stops while initialising USB devices by the look of it.

I've tried again the command with -f and also tried with specifying the optional label when prompted to "RASPIFIRM" but then when I boot it says it can't find the firmware. Maybe I should have tried "RASPIROOT" (these are the existing partition labels on the USB drive).

All of this works perfectly when I use Raspberry Pi OS. I can just run rpi-clone, choose the SD card and then unplug my boot USB, reboot and then it will boot from the SD card. This is great so I always have a "live" bootable, known working image to boot from if any updates or config changes to Home Assistant break anything. Since my home relies on home automation using Home Assistant quite a bit, it really is a great ability to be able to create a bootable snapshot image quickly.

I'd use Raspberry Pi OS under normal circumstances obviously but Home Assistant runs as "unsupported" on Raspberry Pi OS.

If it's impossible to get rpi-clone working reliably on Debian on my Pi4, I might end up having to just use Raspberry Pi OS.

buldezir, 2021-06-29:
> I'd use Raspberry Pi OS under normal circumstances obviously but Home Assistant runs as "unsupported" on Raspberry Pi OS.
> 
> If it's impossible to get rpi-clone working reliably on Debian on my Pi4, I might end up having to just use Raspberry Pi OS.

Why dont u use dockerized version of HA then?
I have it and Unifi Controller running smooth in docker (on Pi4)

diagonali, 2021-06-30:
> > I'd use Raspberry Pi OS under normal circumstances obviously but Home Assistant runs as "unsupported" on Raspberry Pi OS.
> > If it's impossible to get rpi-clone working reliably on Debian on my Pi4, I might end up having to just use Raspberry Pi OS.
> 
> @diagonali Why dont u use dockerized version of HA then?
> I have it and Unifi Controller running smooth in docker (on Pi4)

Interesting. The reason mostly is that I use quite a few of the Addons and as far as I know, you can't get Supervisor to run using the dockerized version and I've never tried to setup Home Assistant Core and then all the Addons I use manually. I'll look it up though.

Real shame I can't get rpi-clone working on Debian though as it's literally the last step to having a fully "supported" install which I can then clone to a live, bootable image which would be much simpler than running HA Core in Docker and doing a manual setup.

ioogithub, 2022-04-20:
Thank you @diagonali for this comment. I was able to successfully boot a clone using your instructions:
> @diagonali had same problem cloning ubuntu server from bootable usb to sd card.
> so the solution is to clone mentioned repo AND switch branch, the fix is not in master!

Now I have a new problem. I am trying to update my clone after the initial clone. This is @billw2's second use case:
> 2) Subsequent clone to the same SD card in USB card reader as example 1

When I run
`./rpi-clone sdb`

I get an error:
```
Destination disk sdb is the booted disk. Cannot clone!
```
This is definitely not the case as I booted from the RPIs built in sd card reader `"/dev/mmcblk0p1"` and the destination is a usb drive at `/dev/sdb`. For some reason the code thinks the usb clone is the drive I booted from. I tried looking at the code to see how rpi-clone was making this determination but I couldn't without success. Could this issue be caused because I used the code in the new branch?

Is there any way to update my clone after the initial copy? Having an updated clone is my ultimate goal with rpi-clone.

diagonali, 2022-05-15:
> Thank you [buldezir](https://github.com/buldezir) for this comment. I was able to successfully boot a clone using your instructions:
> 
> > @diagonali had same problem cloning ubuntu server from bootable usb to sd card.
> > so the solution is to clone mentioned repo AND switch branch, the fix is not in master!
> 
> Now I have a new problem. I am trying to update my clone after the initial clone. This is @billw2's second use case:
> 
> > 2. Subsequent clone to the same SD card in USB card reader as example 1
> 
> When I run `./rpi-clone sdb`
> 
> I get an error:
> 
> ```
> Destination disk sdb is the booted disk. Cannot clone!
> ```
> 
> This is definitely not the case as I booted from the RPIs built in sd card reader `"/dev/mmcblk0p1"` and the destination is a usb drive at `/dev/sdb`. For some reason the code thinks the usb clone is the drive I booted from. I tried looking at the code to see how rpi-clone was making this determination but I couldn't without success. Could this issue be caused because I used the code in the new branch?
> 
> Is there any way to update my clone after the initial copy? Having an updated clone is my ultimate goal with rpi-clone.

The error `Destination disk sdb is the booted disk. Cannot clone!` suggests that you've left both devices plugged in and it's actually booting from the disk you want to copy to. I've done it accidentally myself. Try removing the drive you want to clone to during boot and plugging it in only after. If that's not the issue, I have no idea what could be causing the problem.

andreyl9, 2022-06-28:
Hi, I have same problem, debian, HA with docker, but also didn't work correctly, when I run
`sudo ./rpi-clone sda -l -f -e sda
`
after copying it's didn't whant change `/etc/fstab` and `/boot/cmdline.txt`
`console=tty0 console=ttyS1,115200 root=/dev/mmcblk1p2 rw fsck.repair=yes net.ifnames=0  rootwait 
`

```
# The root file system has fs_passno=1 as per fstab(5) for automatic fsck.
LABEL=RASPIROOT / ext4 rw 0 1
# All other file systems have fs_passno=2 as per fstab(5) for automatic fsck.
LABEL=RASPIFIRM /boot/firmware vfat rw 0 2
```
need to migrate from SD card to SSD, but if I clone it  - after boot from USB receive errors: 

```
Begin: Watching for root file system...
Begin: Running /scripts/local-block ... done
[35.807126] vcc-sd: disabling
Gave up waiting for root file system device. Common problems:
- Boot args (cat /proc/cmdline)
  - check rootdelay= (did the system wait long enough?)
  - Missing modules (cat /proc/modules; ls /dev)
ALERT! /dev/mmcblk1p2 does not exist. Dropping to a shell!
```

I try change to `root=/dev/sda2` and `root=/dev/sda1`  - problem are present...

BTW, if clone from original RPI4 image - working fine - also working under piclone GUI utility...  need your help.. please...

MaxWinterstein, 2022-08-08:
> git clone https://github.com/matthijskooijman/rpi-clone.git
> cd rpi-clone
> git checkout keep-fs-labels

This is way too hard to find here! Was stuck with 

**Press enter for maintenance (or type control-d to continue)**

And some half booting system. Debian onto some usb-hdd and migration to usb-hdd (argon M.2 SATA SSD case).
`/` was available but `/boot/firmware` was not mounted, which caused werid broken boots. Using mentioned branch fixed it for me. Thanks!