billw2/rpi-clone #123: Error with NVME on RPi Compute Module 4

rubberychicken, 2021-07-06:
Hi,

I am getting this when trying to close to a NVME SSD:

pi@raspberrypi:~ $ sudo sys-clone nvme0n1 -f2

  Target disk nvme0n1 ends with a digit so may be a partition.
  sys-clone requires disk names like 'sda' and not partition names like 'sda1'.

Continue anyway?  (yes/no): yes

Booted disk: mmcblk0 31.9GB                Destination disk: nvme0n1 256.1GB
---------------------------------------------------------------------------
Part      Size    FS     Label           Part   Size    FS     Label
1 /boot   256.0M  fat32  --              1      256.0M  fat32  --
2 root     29.5G  ext4   rootfs          2      238.2G  --     --
---------------------------------------------------------------------------
== Initialize: IMAGE partition table - forced by option ==
1 /boot               (48.0M used)   : MKFS  SYNC to nvme0n11
2 root                (1.8G used)    : RESIZE  MKFS  SYNC to nvme0n12
---------------------------------------------------------------------------
-f2                    : force initialize to first two partitions only.
Run setup script       : no.
Verbose mode           : no.
-----------------------:
** WARNING **          : All destination disk nvme0n1 data will be overwritten!
-----------------------:

Initialize and clone to the destination disk nvme0n1?  (yes/no): yes
Optional destination ext type file system label (16 chars max): gateway

Initializing
  Imaging past partition 1 start.
  => dd if=/dev/mmcblk0 of=/dev/nvme0n1 bs=1M count=8 ...
  Resizing destination disk last partition ...
    Resize success.
  Changing destination Disk ID ...
  => mkfs -t vfat -F 32  /dev/nvme0n11 ...
  => mkfs -t ext4 -L gateway /dev/nvme0n12 ...

Syncing file systems (can take a long time)
Syncing mounted partitions:
  Mounting /dev/nvme0n12 on /mnt/clone
mount: /mnt/clone: special device /dev/nvme0n12 does not exist.
    Mount failure of /dev/nvme0n12 on /mnt/clone.
Aborting!

mike-pisman, 2021-08-16:
Hi, I have the same issue. Have you found the solution? 
```ubuntu@ubuntu:~$ sudo rpi-clone nvme0n1

  Target disk nvme0n1 ends with a digit so may be a partition.
  rpi-clone requires disk names like 'sda' and not partition names like 'sda1'.

Continue anyway?  (yes/no): yes

Booted disk: mmcblk0 8.1GB                 Destination disk: nvme0n1 500.1GB
---------------------------------------------------------------------------
Part               Size    FS     Label           Part   Size  FS  Label
1 /boot/firmware   256.0M  fat32  --
2 root               7.3G  ext4   writable
---------------------------------------------------------------------------
== Initialize: IMAGE partition table - partition number mismatch: 2 -> 0 ==
1 /boot/firmware      (126.0M used)  : MKFS  SYNC to nvme0n11
2 root                (3.3G used)    : RESIZE  MKFS  SYNC to nvme0n12
---------------------------------------------------------------------------
Run setup script       : no.
Verbose mode           : no.
-----------------------:
** WARNING **          : All destination disk nvme0n1 data will be overwritten!
-----------------------:

Initialize and clone to the destination disk nvme0n1?  (yes/no): yes
Optional destination ext type file system label (16 chars max): NVME-EXT

Initializing
  Imaging past partition 1 start.
  => dd if=/dev/mmcblk0 of=/dev/nvme0n1 bs=1M count=5 ...
  Resizing destination disk last partition ...
    Resize success.
  Changing destination Disk ID ...
  => mkfs -t vfat -F 32  /dev/nvme0n11 ...
  => mkfs -t ext4 -L NVME-EXT /dev/nvme0n12 ...

Syncing file systems (can take a long time)
Syncing mounted partitions:
  Mounting /dev/nvme0n12 on /mnt/clone
mount: /mnt/clone: special device /dev/nvme0n12 does not exist.
    Mount failure of /dev/nvme0n12 on /mnt/clone.
Aborting!

psistwu, 2021-09-04:
[Update]
Finally, I had my solution tested with the following configuration:
* Hardware: Waveshare Mini Base Board (A) + rpi compute module 4 lite (2Gb ram w/o wifi) + WD SN530 256GB
* System: Manjaro-ARM-minimal-rpi4-21.08

With the correction, everything works quite well. I can now boot my rpi with NVMe drive.

NajibNour, 2021-11-05:
Hey hey! Im facing a similar issue. Im coping over from source "nvme0n1" to a "sda". The error I receive is "Error: Could not stat device /dev/nvme0n1p - No such file or directory." Seems like its messing up the partition names and moving "p1" to "1p". Could this be the same fix but for the source drive?

lsblk output:
`sda           8:0    0 465.8G  0 disk 
└─sda1        8:1    0 465.8G  0 part 
nvme0n1     259:0    0 465.8G  0 disk 
├─nvme0n1p1 259:1    0   512M  0 part /boot/efi
└─nvme0n1p2 259:2    0 465.3G  0 part /`

psistwu, 2021-11-12:
I guess you can try to modify line 571--584 of `rpi-clone` using similar approach.

NajibNour, 2021-11-12:
Yes this is exactly what i ended up doing! Had some issues with UUID's as this is on a ubuntu server but ended up resolving that with the following:
Then you have to edit "/etc/fstab" on the right volume with gedit, nano or else.

In a terminal, type:

`sudo vim /mnt/clone/etc/fstab`
and replace the old UUIDs by the new ones.

WARNING : Modifying "/boot/grub/grub.cfg" is very tricky. It should normally be generated with the command `sudo update-grub`.

Replace the old UUIDs by the new ones in "/boot/grub/grub.cfg" by entering the following in a terminal :

`sudo vim /mnt/clone/boot/grub/grub.cfg`

`:%s/OLDUUID/NEWUUID/gc`

If you use the tricky solution, I recommend you to launch `sudo update-grub` once you have successfully booted the system.

mgrouch, 2021-12-21:
@psistwu 

> [Update] Finally, I had my solution tested with the following configuration:
>
>     * Hardware: Waveshare Mini Base Board (A) + rpi compute module 4 lite (2Gb ram w/o wifi) + WD SN530 256GB
>
>     * [...]

I did the same with raspberry pi buster. It copied file systems fine. But it doesn't boot from NVMe.
All up to date raspberry pi OS (armhf buster) and firmware.

Thanks

psistwu, 2022-01-01:
What is your hardware configuration?

Also, did you change the boot order with [`usbboot`](https://github.com/raspberrypi/usbboot)? You may refer to the following YouTube video for the detail.
https://youtu.be/4Womn10v71s

mgrouch, 2022-01-01:
@psistwu 
CM4 with McuZone board.

chaisaeng, 2022-09-14:
> I think I have a solution to this issue. Here I highlight some lines of the return message of `rpi-clone`:
> 
> ```
> ....
> 
> Syncing file systems (can take a long time)
> Syncing mounted partitions:
>   Mounting /dev/nvme0n12 on /mnt/clone
> mount: /mnt/clone: special device /dev/nvme0n12 does not exist.
>     Mount failure of /dev/nvme0n12 on /mnt/clone.
> Abouting!
> ```
> 
> The message above indicates that `rpi-clone` does not generate correct partition names for NVMe drive, which should be `/dev/nvme0n1p1` and `/dev/nvme0n1p2`. In such a case, one can try the following modification in `rpi-clone`:
> 
> ```shell
> # begin at line 1060
> if [[ ${chk_disk: -1} =~ ^[0-9]$ ]]
> then
> 	if [[ $dst_disk == *"mmcblk"* || $dst_disk == *"nvme"* ]]    # update this line
> 	then
> 		SD_slot_dst=1
>                 ....
> ```
> 
> Unfortunately, I do not have an NVMe drive by my hand so that I cannot really test it. Can somebody verify if the solution works?
> 
> Beware that even the solution above works, it is not a rigorous one. If anyone can improve the solution, it will be much appreciated.

Your Solution is absolutely correct.

geerlingguy, 2022-09-21:
Opened a PR with the two changes discussed above: https://github.com/billw2/rpi-clone/pull/147

jaballarin, 2024-05-04:
> Hey hey! Im facing a similar issue. Im coping over from source "nvme0n1" to a "sda". The error I receive is "Error: Could not stat device /dev/nvme0n1p - No such file or directory." Seems like its [...]

I have the same configuration:
sda 8:0 0 465.8G 0 disk
└─sda1 8:1 0 465.8G 0 part
nvme0n1 259:0 0 465.8G 0 disk
├─nvme0n1p1 259:1 0 512M 0 part /boot/efi
└─nvme0n1p2 259:2 0 465.3G 0 part /

and it gives me the same error:
"Error: Could not stat device /dev/nvme0n1p - No such file or directory."
I have tried to edit the /mnt/clone/etc/fstab file, but I can't find it.

framps, 2024-05-04:
This repository is no longer maintained. Please head over to [this](https://github.com/geerlingguy/rpi-clone/) repository and make sure you use the latest code from there.

jaballarin, 2024-05-04:
> This repository is no longer maintained. Please head over to [this](https://github.com/geerlingguy/rpi-clone/) repository and make sure you use the latest code from there.

Thanks a lot!

I will do so.
Best regards.