billw2/rpi-clone #126: Feature requested - Clone into .img file

thalesmaoa, 2021-08-06:
Nice work, but I was wondering. Is it possible to clone the img?
I have network share, which I can schedule img backups. If I get any corruption, I can easily restore.

TotalCoolness, 2021-08-22:
+1

I would like this also for the same reason.

p1r473, 2021-09-07:
#114

glotim, 2023-07-03:
I am all for this also.

Just upgraded to booting to a USB NVMe drive thanks to your awesome rpi-clone utility. The drive is much larger and faster than the original SD card. Then the question becomes how do I backup the larger SSD as I am filling up the space? Preferably something that can fire off in the middle of the night when nothing else is going on and save it to a network drive. I have a few options and none of them are ideal:
A. Use rsync to backup the important stuff to a network drive. A compressed full image would be much better.
B. Buy a larger SD card or second USB drive just as big as the one I have and use rpi-clone to that drive. For versioning I would have to buy several more SSD's and rotate them. Also not an automated process.
C. Remove the NVMe card from my new Argon One case and put it into an external USB case. Then plug it into another computer and do the "dd" thing then use pishrink to get it down to a more reasonable size and store that file to the backup network drive. Also too much a manual process.

So what do you think? Use a small subset of the rpi-clone code and create rpi_img_backup?

thalesmaoa, 2023-07-03:
This already exists.  Sorry for not posting the solution.

https://github.com/tom-2015/imgclone

closed 2023-07-03 by thalesmaoa