billw2/rpi-clone #131: clone mmcblk0 to sdb while booted with sda

Barabba11, 2021-12-08:
Hi, I've a SD wiht a license I want to clone for a spare copy, I can't boot with mmcblk and install rpi-clone. So I booted with an usb slick with original raspbian, sda. Now if I launch sudo I can just clone sda to mmcblk, as sda is the sytem and I don't want to risk that because I need to preserve. I want to clone mmcblk to sdb, while sda keeps as the root folder. How to do that? Thank you

bwims, 2022-03-22:
Can I bump this please? I'd like to use my a400 to clone two separate devices too.

Barabba11, 2022-09-29:
this question still not solved, any help?
How can I clone mmcblk to sdb (and viceversa) while having sda as raspbian boot drive with rpi-clone?
Looks so easy but I'm getting frustrated

bwims, 2022-09-30:
I solved my own problem later, when I realised that rpi-clone is simply based on the linux "dd" command.

I found a useful article [here](https://www.cyberpunk.rs/clone-micro-sd-card) which explained how it works.

The good thing about rpi-clone is that it backs up the running system without much thought, which is normally what I want to do, when running off an SDD.

If, however, I want to back up a 2nd device to a 3rd device, then this is what I do:

`sudo dd if=/dev/sdX of=/dev/sdY bs=4096 conv=notrunc,noerror `

where if=input (X is the 2nd controller letter), of=output (Y is the 3rd controller letter), 
notrunc = do not truncate the output file
and
noerror = continue after read errors

For my purposes, copying one sd card to another, this works fine.

Barabba11, 2022-10-01:
this question still not solved, any help?
How can I clone mmcblk to sdb (and viceversa) while having sda as raspbian boot drive with rpi-clone?
This program sadly misses the chance to specify the source drive, if it's possible to implement it we can have chances to clone different drives, not only the boot one as source.
Thanks

bwims, 2022-10-01:
Yes, I used to use win32diskimage as you say, and abandoned it the moment I discovered rpi-clone, and through it, dd .

dd has never caused me problems, and it's easy to use.