billw2/rpi-clone #136: Installed to bin instead of sbin

nmbrg, 2022-01-17:
managed to miss that it should be installed to /usr/local/sbin and installed it to /usr/local/bin instead.
Will i have any issues running/using the script/doing backups? or will everything work fine even tho its in the bin folder instead of sbin?

elementzonline, 2023-09-09:
This will work fine. I normally install this to /usr/local/bin as the sbin is not preset in the system path variable PATH.  I am using Debian for RPI version 12.

nmbrg, 2023-09-09:
> This will work fine. I normally install this to /usr/local/bin as the sbin is not preset in the system path variable PATH. I am using Debian for RPI version 12.

Lol, only eight month later. :)
I’m no longer using rpi-clone, but thanks for the info.