billw2/rpi-clone #142: rpi-clone won't initialize destination when source is bullseye

Dan-Essin, 2022-08-13:
Hello,
I have upgraded my rpi to bullseye, For some reason this is the layout of the sdcard:
sdf      8:80   1  29.8G  0 disk 
├─sdf1   8:81   1   1.6G  0 part 
├─sdf2   8:82   1     1K  0 part 
├─sdf5   8:85   1    32M  0 part /run/media/essin/SETTINGS
├─sdf6   8:86   1    69M  0 part /run/media/essin/boot
└─sdf7   8:87   1    28G  0 part /run/media/essin/root

rpi-clone chokes on this, complains about not being able to create partition 6. Then it stops.
I was under the impression that msdos format only allowed 4 primary partitions.
Can anyone explain/fix this?

Thank you

framps, 2022-08-13:
Looks ugly. What's the output of `blkid` and `mount`?

Anyhow I think it's not a good idea to upgrade. It's not officially supported any you may get a lot of issues with the upgrade. I suggest to install the new OS version and migrate all applications.

Dan-Essin, 2022-08-14:
I have done a bit of work on this. It turns out that this sdcard was originally a NOOBS distro. That's where all the extra partitions came from. I didn't delve into the code that parses the disk layout to decide if this anomaly could have been detected but I have fixed it.

I dd'ed the two actual rpi partitions to img files. Then I reformatted the card, recreated the two partitions so that they were sdf1 and sdf2, then dd'ed the img's back to the card.

After adjusting fstab and cmdline.txt everything is now working fine.

arigit, 2022-10-05:
@Dan-Essin  what errors did you get? Trying to initialize a new card with rpi-clone, in bullseye, and seeing this,

```
=> rsync /boot/ /mnt/clone/boot  ...rsync: [generator] chown "/mnt/clone/boot/." failed: Operation not permitted (1)
rsync: [generator] readlink_stat("/mnt/clone/boot/.Spotlight-V100") failed: Input/output error (5)
rsync: [generator] readlink_stat("/mnt/clone/boot/.fseventsd") failed: Input/output error (5)
rsync: [generator] readlink_stat("/mnt/clone/boot/dtb-5.10.22-4-osmc") failed: Input/output error (5)
rsync: [generator] readlink_stat("/mnt/clone/boot/overlays") failed: Input/output error (5)
rsync: [generator] readlink_stat("/mnt/clone/boot/dtb-5.10.32-1-osmc") failed: Input/output error (5)

```

not sure what to do.