billw2/rpi-clone #144: RPI CM4 clone from eMMC to NVMe drive fails due to partition naming

tkuehne71, 2022-08-29:
I have a Raspberry Compute module 4  running Ubuntu server 22.04 LTS from the internal eMMC flash of the CM4. I am trying to clone from the eMMC flash memory of the compute module to an NVMe drive that is attached to the PCIe slot and subsequently boot and run the system from the NVMe drive.

The device is enumerated as "/dev/nvme0n1", and the two partitions would appear as "/dev/nvme0n1p1" and "/dev/nvme0n1p2". 

Rpi-clone does not appear to be able to support this naming, since it only wants to add a single digit number for the partitions, i.e. it is trying to name the partition as "/dev/nvme0n11" or "/dev/nvme0n12". 

Is it possible to explicitly name the output partitions to support this naming scheme?

Here's the output of the failed run:

` rpi-clone /dev/nvme0n1 -v

  Target disk nvme0n1 ends with a digit so may be a partition.
  rpi-clone requires disk names like 'sda' and not partition names like 'sda1'.

Continue anyway?  (yes/no): yes

Booted disk: mmcblk0 31.3GB                Destination disk: nvme0n1 128.0GB
---------------------------------------------------------------------------
Part               Size    FS     Label           Part   Size    FS     Label
1 /boot/firmware   256.0M  fat32  --              1      256.0M  fat32  --
2 root              28.9G  ext4   writable        2      119.0G  --     --
---------------------------------------------------------------------------
== Initialize: IMAGE partition table - FS types conflict ==
1 /boot/firmware      (73.0M used)   : MKFS  SYNC to nvme0n11
2 root                (3.7G used)    : RESIZE  MKFS  SYNC to nvme0n12
---------------------------------------------------------------------------
Run setup script       : no.
Verbose mode           : yes.
-----------------------:
** WARNING **          : All destination disk nvme0n1 data will be overwritten!
-----------------------:

Initialize and clone to the destination disk nvme0n1?  (yes/no): yes
Optional destination ext type file system label (16 chars max):

Initializing
  Imaging past partition 1 start.
  => dd if=/dev/mmcblk0 of=/dev/nvme0n1 bs=1M count=5 ...
  Resizing destination disk last partition ...
    Resize success.
  Changing destination Disk ID ...
  => mkfs -t vfat -F 32  /dev/nvme0n11 ...
  => mkfs -t ext4  /dev/nvme0n12 ...

Syncing file systems (can take a long time)
Syncing mounted partitions:
  Mounting /dev/nvme0n12 on /mnt/clone
mount: /mnt/clone: special device /dev/nvme0n12 does not exist.
    Mount failure of /dev/nvme0n12 on /mnt/clone.
Aborting!
`

tkuehne71, 2022-08-29:
Duplicate of #123, apologies

closed 2022-08-29 by tkuehne71

framps, 2022-08-29:
FYI

[raspiBackup](https://www.linux-tips-and-tricks.de/en/all-raspibackup-articles/612-support-for-raspberry-compute-module-4-and-nvme/) recently got support for CM4 ;-)