billw2/rpi-clone #150: rpi-clone from NVME to USB-Stick

Marxi2703, 2023-01-12:
I have a Notebook with an NVME SSD and Debian 11 64 bit.

If I install rpi-clone 

$ git clone https://github.com/billw2/rpi-clone.git 
        $ cd rpi-clone
        $ sudo cp rpi-clone /usr/local/sbin/sys-clone
        $ sudo cp rpi-clone-setup /usr/local/sbin/sys-clone-setup

I get no Errors.

If I want to Clone my System from the nvme ssd I get the following error:

root@debian:~# rpi-clone
Error: Could not stat device /dev/nvme0n1p - No such file or directory.

sdc           8:32   1  28.8G  0 disk
nvme0n1     259:0    0 238.5G  0 disk
├─nvme0n1p1 259:1    0   512M  0 part /boot/efi
└─nvme0n1p2 259:2    0   237G  0 part /

Sdc is my USB Stick where I have to Clone my System
Nvme01n is the System and boot Partition.

Can anyone help me with this error?
I want with rpi-clone a bootable system if my system crash.

Thanks in advance

Marxi2703, 2023-01-13:
Sorry don't solve my problem.
I changed in rpi-clone the Line 
# begin at line 1060
if [[ ${chk_disk: -1} =~ ^[0-9]$ ]]
then
	if [[ $dst_disk == *"mmcblk"* || $dst_disk == *"nvme"* ]]    # update this line
	then
		SD_slot_dst=1
                ....
But I get the same error.
My problem is that I want to clone from nvme to USB-Stick not the other way.
Can anyone help me?

Marxi2703, 2023-01-13:
My NVME has 256 GB but I only have written 20 GB
My destination has 128 GB.
Must the destination 256 GB or larger?

framps, 2023-01-13:
> Must the destination 256 GB or larger?

As far as I know it has to have at least 20GB and some space for the boot partition.

> Can anyone help me?

I don't have a NVMe SSD but a CM4 with NVMe. I will check whether #147 fails for me.

Marxi2703, 2023-01-13:
OK I try to copy with this command
dd if=/dev/nvme0n1 of=/dev/sdc bs=100M status=progress
My NVME has 256 GB and my destination device 128 GB 
after 128 GB I get this error 
dd: error writing '/dev/sdc': No space left on device
So I think he will copy the whole source device not the used space.

framps, 2023-01-13:
This will not work. dd is a 1:1 copy and you  need at least the same size of your source device for your target device. 

I thought you want to use rpi-clone :worried:

Marxi2703, 2023-01-13:
Yes I want to use rpi-clone but this dont work for me.
I want to explain it:
I install rpi-clone with these commands:
$ git clone https://github.com/billw2/rpi-clone.git 
	$ cd rpi-clone
	$ sudo cp rpi-clone /usr/local/sbin/sys-clone
	$ sudo cp rpi-clone-setup /usr/local/sbin/sys-clone-setup

Then I run
root@debian:~/rpi-clone# rpi-clone
Error: Could not stat device /dev/nvme0n1p - No such file or directory.
This is my problem

I can run this:
root@debian:~/rpi-clone# sudo bash -x rpi-clone sdc
+ version=2.0.22
+ export PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
+ PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
+ grub_auto=1
++ basename rpi-clone
+ PGM=rpi-clone
+ setup_command=rpi-clone-setup
+ rsync_options='--force -rltWDEHXAgoptx'
++ id -u
+ '[' 0 '!=' 0 ']'
+ raspbian=0
+ raspbian_buster=0
+ '[' -f /etc/os-release ']'
++ cat /etc/os-release
++ grep PRETTY
+ pretty='PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"'
+ [[ PRETTY_NAME="Debian GNU/Linux 11 (bullseye)" == *\R\a\s\p\b\i\a\n* ]]
+ (( raspbian ))
+ commands='rsync parted fdisk findmnt column fsck.vfat'
+ packages='rsync parted util-linux mount bsdmainutils dosfstools'
+ need_packages=
+ idx=1
+ for cmd in $commands
+ command -v rsync
+ (( ++idx ))
+ for cmd in $commands
+ command -v parted
+ (( ++idx ))
+ for cmd in $commands
+ command -v fdisk
+ (( ++idx ))
+ for cmd in $commands
+ command -v findmnt
+ (( ++idx ))
+ for cmd in $commands
+ command -v column
+ (( ++idx ))
+ for cmd in $commands
+ command -v fsck.vfat
+ (( ++idx ))
+ '[' '' '!=' '' ']'
+ clone=/mnt/clone
+ clone_src=/mnt/clone-src
+ clone_log=/var/log/rpi-clone.log
++ hostname
+ HOSTNAME=debian
++ findmnt /boot -o source -n
++ grep /dev/
+ src_boot_dev=
++ findmnt / -o source -n
++ grep /dev/
+ src_root_dev=/dev/nvme0n1p2
+ SD_slot_boot=0
+ SD_slot_dst=0
+ src_part_base=
[... 248 lines cut ...]
+ (( p <= n_src_parts ))
+ (( !src_exists[p] ))
+ blocks=0
+ dst_mount_flag[p]=0
+ '[' / '!=' '' ']'
+ dst_mount_flag[p]=1
++ echo '/dev/nvme0n1p2 243520884 16104488 214973364 7% /
/dev/nvme0n1p1 523248 5904 517344 2% /boot/efi'
++ grep -m 1 '^/dev/nvme0n1p2'
++ cut -d ' ' -f 3
+ blocks=16104488
+ src_used_sectors[p]=32208976
+ (( p++ ))
+ (( p <= n_src_parts ))
+ '[' sdc = '' ']'
++ cat /proc/partitions
++ grep -m 1 sdc
+ chk_disk='   8       32  125034840 sdc'
+ '[' '   8       32  125034840 sdc' == '' ']'
+ dst_part_base=sdc
+ [[ c =~ ^[0-9]$ ]]
+ '[' nvme0n1 == sdc ']'
++ parted -m /dev/sdc unit s print
++ tr -d ';'
Error: Invalid argument during seek for read on /dev/sdc

wptracy, 2023-02-20:
I tried this PR on my orange pi 5 running armbian Ubuntu 22.04.2 LTS desktop.

Can you make this work on my orangepi5.

```
wptracy@orangepi5:~/Downloads/rpi-clone$ sudo ./rpi-clone -v sda

  Destination disk partition /dev/sda1 is mounted on /media/wptracy/BC1B-131B.
  The clone cannot proceed unless it is unmounted.
Do you want to unmount /media/wptracy/BC1B-131B?  (yes/no): yes

Booted disk: nvme0n1 128.0GB               Destination disk: sda 31.9GB
---------------------------------------------------------------------------
Part      Size    FS     Label           Part   Size    FS     Label  
1 /boot   256.0M  fat16  --              1       29.7G  fat32  --     
2 root     29.2G  ext4   armbi_root                                   
---------------------------------------------------------------------------
== Initialize: IMAGE partition table - partition number mismatch: 2 -> 1 ==
1 /boot               (152.5M used)  : MKFS  SYNC to sda1
2 root                (5.1G used)    : RESIZE  MKFS  SYNC to sda2
---------------------------------------------------------------------------
Run setup script       : no.
Verbose mode           : yes.
-----------------------:
** WARNING **          : All destination disk sda data will be overwritten!
-----------------------:

Initialize and clone to the destination disk sda?  (yes/no): yes
Optional destination ext type file system label (16 chars max): 

Initializing
  Imaging past partition 1 start.
  => dd if=/dev/nvme0n1 of=/dev/sda bs=1M count=20 ...
  Resizing destination disk last partition ...
    Resize failed.  See /tmp/rpi-clone-output.
    Try running rpi-clone again.

GPT PMBR size mismatch (250069679 != 62333951) will be corrected by write.
wptracy@orangepi5:~/Downloads/rpi-clone$ ls
ls           lsb_release  lsipc        lsmem        lsns         lspgpot
lsattr       lscpu        lslocks      lsmod        lsof         lsusb
lsblk        lsinitramfs  lslogins     lsmtd        lspci        
wptracy@orangepi5:~/Downloads/rpi-clone$ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
sda           8:0    1  29.7G  0 disk 
mtdblock0    31:0    0    16M  0 disk 
zram0       254:0    0   3.8G  0 disk [SWAP]
zram1       254:1    0    50M  0 disk /var/log
nvme0n1     259:0    0 119.2G  0 disk 
├─nvme0n1p1 259:1    0   256M  0 part /boot
└─nvme0n1p2 259:2    0  29.2G  0 part /var/log.hdd
                                      /
wptracy@orangepi5:~/Downloads/rpi-clone$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.2 LTS
Release:	22.04
Codename:	jammy
wptracy@orangepi5:~/Downloads/rpi-clone$ 

```