billw2/rpi-clone #151: Recent RPi OS version aren't correctly detected

martignoni, 2023-01-15:
Logic to detect RPi OS version has changed and should be taken in account.

framps, 2023-01-15:
Added a fix it on [my branch](https://github.com/framps/rpi-clone)

martignoni, 2023-01-15:
See https://github.com/billw2/rpi-clone/pull/152 (more generic fix IMHO).

J-Luc16, 2023-02-13:
I am surprised, this is what is returned by a raspbian Bullseye :

cat /etc/os-release

PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye

RPI_CLONE never stopped working for me, also with the bullseye.

J-Luc16, 2023-02-13:
Built very recently (january). I understand your concern about the 32/64 bits, I am not skilled enough to give my advice.
FYI it's the latest kernel (6.1), I don't know if it matters in the answer to _os-release_.
Thanks for all.