billw2/rpi-clone #153: RPi4 SD card to USB Drive failed - drive unmounted

mpikosdimitris, 2023-01-23:
Hello, i have this issue, trying to clone my sd card to an ssd drive with usb case and i'm getting my drive unmounted and not recognized by lsblk until i restart the system. 
```
pi@raspberrypi:~ $ **sudo lsblk**
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
**sda           8:0    0 111.8G  0 disk
└─sda1        8:1    0 111.8G  0 part**
mmcblk0     179:0    0  29.7G  0 disk
├─mmcblk0p1 179:1    0   256M  0 part /boot
└─mmcblk0p2 179:2    0  29.5G  0 part /

pi@raspberrypi:~ $ **sudo rpi-clone sda**

Booted disk: mmcblk0 31.9GB                Destination disk: sda 120.0GB
---------------------------------------------------------------------------
Part      Size    FS     Label           Part   Size    FS    Label
1 /boot   256.0M  fat32  --              1      111.8G  ntfs  --
2 root     29.5G  ext4   rootfs
---------------------------------------------------------------------------
== Initialize: IMAGE partition table - partition number mismatch: 2 -> 1 ==
1 /boot               (49.5M used)   : MKFS  SYNC to sda1
2 root                (13.9G used)   : RESIZE  MKFS  SYNC to sda2
---------------------------------------------------------------------------
Run setup script       : no.
Verbose mode           : no.
-----------------------:
** WARNING **          : All destination disk sda data will be overwritten!
-----------------------:

Initialize and clone to the destination disk sda?  (yes/no): yes
Optional destination ext type file system label (16 chars max):

Initializing
  Imaging past partition 1 start.
  => dd if=/dev/mmcblk0 of=/dev/sda bs=1M count=8 ...
  Resizing destination disk last partition ...
    Resize failed.  See /tmp/rpi-clone-output.
    Try running rpi-clone again.

fdisk: cannot open /dev/sda: No such file or directory
```
this is the part that rpi-clone fails and after that the lsblk not recognizes the drive.

```
pi@raspberrypi:~ $ sudo lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
mmcblk0     179:0    0 29.7G  0 disk
├─mmcblk0p1 179:1    0  256M  0 part /boot
└─mmcblk0p2 179:2    0 29.5G  0 part /
pi@raspberrypi:~ $
```
this is the outpout file
```
pi@raspberrypi:/tmp $ cat rpi-clone-output
sfdisk: cannot open /dev/sda: No such file or directory

====rpi-clone
==orig:
label: dos
label-id: 0x6707a26f
device: /dev/mmcblk0
unit: sectors
sector-size: 512

/dev/mmcblk0p1 : start=        8192, size=      524288, type=c
/dev/mmcblk0p2 : start=      532480, size=    61801472, type=83

==edited:
label: dos
label-id: 0x6707a26f
device: /dev/mmcblk0
unit: sectors
sector-size: 512

/dev/mmcblk0p1 : start=        8192, size=      524288, type=c
/dev/mmcblk0p2 : start=      532480,  type=83

```

any suggestions ?

ThiloHettmer, 2023-06-01:
I got the same issue. Have you fixed it?

ThiloHettmer, 2023-06-01:
Little update for anyone who encounters the same issue. It was the ssd case that has the jmicron chipset. Look for asmedia instead.

rhys96999, 2023-11-01:
Same problem for me and same solution