billw2/rpi-clone #155: How to clone with new UUID?

mgnthe54, 2023-05-05:
Hello, I am desperately looking for how to clone an SSD (therefore USB) to an SD **with different UUID numbers.**
A bit like the "new partition UUIDs" option in the "SD Card Copier" accessory.
Currently on my pi4 I boot in USB then SD if USB not available (eeprom).
I do a full copy of my SSD to my SD every week using the following command:
sudo rpi-clone mmcblk0 -v -U
in this configuration in case of reboot of the Pi I restart on the SD (although the eeprom of the pi is on USB,) which I do not want, and on the desktop we did not see the SSD!
So I tried this:
sudo rpi-clone -l mmcblk0 -v -U
there I reboot well on the SSD and the SD card is well seen on the desktop but they always have the same UUID and that causes problems for example impossible to succeed in ejecting via the desktop the SD card!

```
pi@raspberrypi:~ $ sudo blkid
/dev/sda1: LABEL_FATBOOT="bootfs" LABEL="bootfs" UUID="6229-BAA7" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="392a8b00-01"
/dev/sda2:LABEL="rootfs" UUID="7c7dfdc9-556c-4551-bb46-391b1dd27cc0" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="392a8b00-02"

/dev/mmcblk0p1: LABEL_FATBOOT="bootfs" LABEL="bootfs" UUID="6229-BAA7" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="3203184d-01"
/dev/mmcblk0p2: LABEL="rootfs" UUID="7c7dfdc9-556c-4551-bb46-391b1dd27cc0" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="3203184d-02"
```

How can I do this? (change UUID of the copy therefore of the SD?)
THANKS

framps, 2023-05-11:
I unfortunately don't know how to achieve this with rpi-clone :-(.

But as an alternative you can achieve this with [raspiBackup](https://www.linux-tips-and-tricks.de/en/backup) :-) Just create a backup of your SSD with raspiBackup and then [restore the SSD backup](https://www.linux-tips-and-tricks.de/en/restore) to your SD card with option `--updateUUIDs`.

mgnthe54, 2023-05-11:
Thanks for that, I'll take a look.
Otherwise I did this (script) to modify the copy.
```
sudo rpi-clone -l mmcblk0 -v -U >> /home/pi/script/copy_disk_boot.log
echo "Cloning complete." >> /home/pi/script/copy_boot_disk.log
echo "">> /home/pi/script/copy_boot_disk.log

echo "blkid clone output:" >> /home/pi/script/copy_boot_disk.log
sudo blkid >> /home/pi/script/copy_boot_disk.log
echo "">> /home/pi/script/copy_boot_disk.log

echo "Change UUID partition 1 (vfat): " >> /home/pi/script/copy_disk_boot.log
sudo mlabel -i /dev/mmcblk0p1 -n >> /home/pi/script/copy_disk_boot.log
echo "">> /home/pi/script/copy_boot_disk.log

sudo blkid >> /home/pi/script/copy_boot_disk.log
echo "">> /home/pi/script/copy_boot_disk.log

echo "Renamed boot disk, label disappeared:" >> /home/pi/script/copy_boot_disk.log
sudo mlabel::"bootfs" -i/dev/mmcblk0p1 >> /home/pi/script/copy_boot_disk.log

echo "">> /home/pi/script/copy_boot_disk.log

sudo blkid >> /home/pi/script/copy_boot_disk.log
echo "">> /home/pi/script/copy_boot_disk.log

echo "Change rootfs disk UUID:" >> /home/pi/script/copy_boot_disk.log

echo "Unmount partition 2:" >> /home/pi/script/copy_boot_disk.log
sudo umount /dev/mmcblk0p2
echo "">> /home/pi/script/copy_boot_disk.log

echo "File system check:">> /home/pi/script/copy_boot_disk.log
sudo e2fsck -fy /dev/mmcblk0p2 >> /home/pi/script/copy_disk_boot.log
echo "">> /home/pi/script/copy_boot_disk.log

echo "Random change UUID partition 2 (ext4):">> /home/pi/script/copy_disk_boot.log
sudo tune2fs -U random /dev/mmcblk0p2 >> /home/pi/script/copy_disk_boot.log
echo "">> /home/pi/script/copy_boot_disk.log
echo "Renamed (if needed) partition 2 (ext4):">> /home/pi/script/copy_disk_boot.log
sudo tune2fs -L ROOTFS /dev/mmcblk0p2 >> /home/pi/script/copy_disk_boot.log
echo "">> /home/pi/script/copy_boot_disk.log

echo "Name visible (blkid) only after reboot except via desktop">> /home/pi/script/copy_disk_boot.log
sudo blkid >> /home/pi/script/copy_boot_disk.log
echo "">> /home/pi/script/copy_boot_disk.log
echo "Changes completed, to make the SD bootable thought of modifying in the SD /boot/cmdline.txt (put the new PARTUUID)." >> /home/pi/script/copy_boot_disk.log

exit 0
```

framps, 2023-05-11:
Your script looks pretty good :+1: 

Would be great if you attach the script file instead to add it in a reply. This makes it much easier to reuse your srcipt ;-)

mgnthe54, 2023-05-11:
Here is the complete script with log file size management!
[script.txt](https://github.com/billw2/rpi-clone/files/11456724/script.txt)